realsenseai/librealsense #10016: realsense-viewer not detecting d435i camera connected to the jetson board. Realsense ros-wrapper not working properly.

Rafi-UST, 2021-12-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | realsense-viewer not detecting camera | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 | NVIDIA Jetson Nano|
| SDK Version                            |   Tried v2.50.0, v2.48.0, v2.49.0, v2.38.0                         | 
| Segment			|  Robotics                           | 

### Issue Description
I have been trying to compile different versions of intel realsense sdk (librealsense v2.50.0 , v2.49.0 , v2.48.0, v2.38.0 etc)from source , but none of the versions (even after successful build )are detecting the realsense camera attached. 

**Realsense viewer is displaying**  : Please connect intel realsense device (for all the versions of librealsense mentioned above except for v2.49.0). Even though its working with v2.49.0 , **my end goal is to get a feed in rviz using the realsense-ros wrapper ( `roslaunch realsense2_camera   rs_camera.launch`).** 

The output attached below is what i got while executing the roslaunch comand (for v2.49.0 of liibrealsense)
[realsense_rviz_camera_launch_output.txt](https://github.com/IntelRealSense/librealsense/files/7631446/realsense_rviz_camera_launch_output.txt)

Tried both jetson hacks method and linux manual installation method for building librealsense from source.  

Also attaching the catkin_ws directory structure (completed catkin_make successfully)
[catkin_ws_directory_structure.txt](https://github.com/IntelRealSense/librealsense/files/7631473/catkin_ws_directory_structure.txt)

Tagging my colleague also with this issue: @iamrajee

MartyG-RealSense, 2021-12-01:
Hi @Rafi-UST   At the bottom of your ROS launch log it states that the RealSense ROS wrapper was built with 2.49.0 but is running with 2.50.0 (which the 2.3.2 wrapper version is designed specifically for - there was not a wrapper version published for 2.49.0).  After the librealsense version is changed (like with your upgrade from 2.49 to 2.50), the RealSense ROS wrapper should be built from source again.  

If you have been having difficulty with building librealsense from source, the installation guide for Nano at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 shared by a RealSense user may be helpful.

Rafi-UST, 2021-12-01:
Hi @MartyG-RealSense , thanks for the quick response. If I update the version of librealsense (to v2.50.0) then the realsense-viewer prompts to connect a realsense device. Also rs-enumerate devices command does not detect the camera attached. I have compiled the ros-wrapper (realsense-ros) inside the catkin workspace as required after compiling librelasense from source.The wrapper is getting successfully compiled, no issues there.

MartyG-RealSense, 2021-12-01:
Ideally, a ROS wrapper version should be matched to a specific librealsense version, but that is not possible with 2.49.0 as it does not have a matching ROS wrapper version.  In those rare circumstances, if 2.49.0 is the only SDK that will work for you then I would suggest trying to build one wrapper version older - version **2.3.1** that was designed for librealsense 2.48.0.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.1

If the option is available to you, I would also recommend a wipe and fresh reinstall of your Nano to ensure that only one librealsense source-code build (2.50.0) is installed on it.

Rafi-UST, 2021-12-01:
Just days before I tried with the ros wrapper version 2.3.1 and librealsense v2.49.0 but the output form roslaunch was no realsense devices detected.

Although now i did a fresh reinstall of librealsense (v2.50.0). The obtained outputs are attached below. 

**roverteam@roverteam-desktop:~$** realsense-viewer 
 01/12 14:19:23,212 WARNING [547620716560] (ds5-factory.cpp:1152) DS5 group_devices is empty.
 01/12 14:19:24,511 WARNING [547620716560] (ds5-factory.cpp:1152) DS5 group_devices is empty.
 01/12 14:19:24,723 WARNING [547620716560] (ds5-factory.cpp:1152) DS5 group_devices is empty.

**roverteam@roverteam-desktop:~$** rs-enumerate-devices 
No device detected. Is it plugged in?

MartyG-RealSense, 2021-12-01:
There was a case of the **DS5 group_devices is empty** error on Nano in https://github.com/IntelRealSense/librealsense/issues/8100 in January 2021.  The solution that worked in that particular case was to do a wipe and then install from source code using the earlier mentioned method at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 - did you use that method to perform the 2.50.0 reinstall, please?

Rafi-UST, 2021-12-01:
I followed the RSUSB backend method to build from source, now relasense-viewer is detecting the camera (Executed the **libvuc_installation.sh script** to do so). I am able to get the feed camera feed in rviz.

The `roslaunch realsense2_camera rs_camera.launch ` is giving the following output: 

roverteam@roverteam-desktop:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/roverteam/.ros/log/1dac3f78-5298-11ec-ba6d-00044be62a16/roslaunch-roverteam-desktop-6176.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://127.0.0.1:43509/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 74 lines cut ...]
[ INFO] [1638357388.920039100]: Stereo Module was found.
[ INFO] [1638357389.067551954]: RGB Camera was found.
[ INFO] [1638357389.068308691]: Motion Module was found.
[ INFO] [1638357389.068490049]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1638357389.068592031]: num_filters: 0
[ INFO] [1638357389.068649845]: Setting Dynamic reconfig parameters.
[ INFO] [1638357393.007057422]: Done Setting Dynamic reconfig parameters.
[ INFO] [1638357393.008463446]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1638357393.011895977]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1638357393.012029105]: setupPublishers...
[ INFO] [1638357393.026860143]: Expected frequency for depth = 30.00000
[ INFO] [1638357393.167077610]: Expected frequency for color = 30.00000
[ INFO] [1638357393.239949796]: setupStreams...
[ INFO] [1638357394.166393022]: SELECTED BASE:Depth, 0
[ INFO] [1638357394.193577693]: RealSense Node Is Up!
[ WARN] [1638357394.282498758]: 
 01/12 16:46:34,284 WARNING [547759296896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/12 16:46:34,335 WARNING [547759296896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/12 16:46:34,386 WARNING [547759296896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/12 16:46:34,436 WARNING [547759296896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/12 16:47:00,718 WARNING [547759296896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

**I hope the last warning is nothing to be concerned of?**

Thanking you in advance @MartyG-RealSense for the quick response and support.

MartyG-RealSense, 2021-12-01:
If the camera topics are being published correctly in RViz (which you can check by running it with **rosrun rviz rviz**) then you can safely ignore the **control_transfer returned** warnings.

Rafi-UST, 2021-12-01:
@MartyG-RealSense,  yes the camera topics are being published properly.(checked raw and depth). 

Thanks !!! for the support.

MartyG-RealSense, 2021-12-01:
You are very welcome - it's excellent to hear that you succeeded  :)

closed 2021-12-01 by Rafi-UST