realsenseai/librealsense #10017: Is it possible to get the distance from the rgb and depth image without using the sdk?

poornimajd, 2021-12-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Operating System & Version |  Ubuntu 18 | 
| Platform                                 | PC.  |
| Segment			|  Robot                         | 

### Issue Description
I have the depth image and the rgb image.But I want to get the distance of a given point in the image.One way is to use the get_distance function using the sdk.But I just have the stream of the images or depth coming in.I want to get the distance of a given point without using a sdk.
I checked out some of the issues in the repo,but could not find a solution.
Any suggestion is greatly appreciated.

MartyG-RealSense, 2021-12-01:
Hi @poornimajd  If you want to obtain distance from images without using an SDK then **OpenCV** would be an appropriate way to do it.  If you are analyzing saved image files such as jpg or png then googling for **opencv depth distance png** should provide relevant search results, whilst if you are using a live camera stream then you could search for **opencv depth distance webcam**

An example of finding distance from a live camera using Python and OpenCV is in the link below.

https://www.pyimagesearch.com/2015/01/19/find-distance-camera-objectmarker-using-python-opencv/

poornimajd, 2021-12-01:
Thank you for the quick reply.I will check out the link.

closed 2021-12-01 by poornimajd