realsenseai/librealsense #10020: TypeError: 'pyrealsense2.pyrealsense2.config' object is not subscriptable

vigkrish123, 2021-12-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Operating System & Version |   windows 10  |  
| Platform                                 | Laptop |
| Language                            |  {python/opencv}                          | 

I am currently using Realsense D435i camera for my project. I am using the pyrealsense2 library for getting the stream but I am facing the error "TypeError: 'pyrealsense2.pyrealsense2.config' object is not subscriptable" every time I try to access the frames from the camera for detection purpose.

I have mentioned the code below to get a better picture:-
`pipeline = rs.pipeline()

        config = rs.config()

        config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

        pipeline.start(config)
        colorwriter = cv2.VideoWriter('output_realsense.avi', cv2.VideoWriter_fourcc(*'XVID'), 30, (640, 480), 1)

        # the main loop
        batch_size  = 1
        images      = []
        video = []
        while True:
            frames = pipeline.wait_for_frames()
            color_frame = frames.get_color_frame()
            color_image = np.asanyarray(color_frame.get_data())
            image = color_image
            ret_val = True
            if ret_val == True: images += [image]

            if (len(images)==batch_size) or (len(images)>0):
                batch_boxes = get_yolo_boxes(infer_model, images, net_h, net_w, config['model']['anchors'], obj_thresh, nms_thresh)
             
                for i in range(len(images)):
                    draw_boxes(images[i], batch_boxes[i], config['model']['labels'], obj_thresh)
                    
                    cv2.imshow('video with bboxes', images[i])
                    colorwriter.write(images[i])

                images = []
            if cv2.waitKey(1) == 27: 
                break  # esc to quit
        cv2.destroyAllWindows()
        colorwriter.release()`

The error occurs when batch_boxes are created (infer_model = loading the Keras model, net_h and net_w = defined value, config values from the config file containing box dimensions and Obj_thresh and nms_thresh are defined values)

If I use my laptop webcam as the source of stream, it works but If i use pyrealsense and realsense camera, I face this issue. Kindly provide me with some insight on what I can do or why its not able to access the frames.

MartyG-RealSense, 2021-12-01:
Hi @vigkrish123   Your pipeline setup and config code looks okay.  As a starting point in investigating your case, I note that you have placed your **colorwriter** instruction directly after the pipe start instruction, whilst a similar .avi saving line in https://github.com/IntelRealSense/librealsense/issues/2731#issuecomment-529938267 places it immediately before the pipe start line.  

Please try moving the colorwriter instruction to the line before pipe start in order to eliminate it as a possible cause of your problem.

```
colorwriter = cv2.VideoWriter('output_realsense.avi', cv2.VideoWriter_fourcc(*'XVID'), 30, (640, 480), 1)
pipeline.start(config)
```

vigkrish123, 2021-12-01:
I tried moving the pipeline start line but I  still get the same error. I don't understand why it says 'pyrealsense2.pyrealsense2.config' object is not subscriptable because normal rgb streaming and saving works fine but when it reaches batch_boxes, it stops

MartyG-RealSense, 2021-12-01:
I have not seen a script like this before.  Since the program is apparently failing at the batch_boxes line though, I wonder whether the error message is referring to having problems with the **config['model']** part of that line (the config values from the config file containing box dimensions) and not the camera stream config.

vigkrish123, 2021-12-01:
I kinda fixed the issue. I used rs.align and align.process and it somehow worked. But facing issue with saving the stream though. Anyways I will close the issue. Thanks a lot for your quick response.

closed 2021-12-01 by vigkrish123

MartyG-RealSense, 2021-12-01:
You are very welcome, @vigkrish123 - it's great to hear that you made progress!