realsenseai/librealsense #10021: Build RealSense SDK for Android OS

18518021156, 2021-12-02:
-------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455  | 
| Firmware Version                   |  | 
| Operating System & Version |  Android  | 
| Kernel Version (Linux Only)    | | 
| Platform                                 | |
| SDK Version                            | 2.49 | 
| Language                            | Android| 
| Segment			|  Smartphone  | 

### Issue Description
Hello,I want to build the RealSense SDK for Android OS through Android Studio, when I went to the fourth step, I had the following error：

A problem occurred evaluating project ':camera'.
> Failed to apply plugin [id 'com.android.internal.version-check']
   > Minimum supported Gradle version is 6.5. Current version is 6.1.1. If using the gradle wrapper, try editing the distributionUrl in E:\DeveloperProjects\RealSenseProject\librealsense-2.49.0\wrappers\android\gradle\wrapper\gradle-wrapper.properties to gradle-6.5-all.zip
[image]
there is no gradle-wrapper.properties file. And I tried to change the Gradle version, but it didn't work.
Do you have any suggestions？thanks.

MartyG-RealSense, 2021-12-02:
When you go to the **File > Open** menu option (as described in Step 4 of the Android wrapper installation instructions), instead of selecting a location on the computer you can paste the website address that Intel provide in their wrapper installation guide into the address bar at the top of the Open pop-up box. That address is:

`https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/build.gradle`

Once this address has been pasted into the address bar and the **OK** button is clicked , the build gradle should be downloaded into Android Studio and installed automatically.

[image]

Do you still receive the same error about the minimum gradle version if the above procedure is followed, please?

18518021156, 2021-12-02:
Thanks for your reply. Why is it that when I paste the link and click OK to not build the project, it jumps to other local projects on my computer.

MartyG-RealSense, 2021-12-02:
Do you have an active internet connection when you click OK, since it is a web link?

18518021156, 2021-12-02:
Yes, I clicked OK on an active network connection

MartyG-RealSense, 2021-12-02:
Thanks very much for your patience.  This appears to be an issue with Android Studio in general and not a RealSense-specific issue.  I carefully researched methods to upgrade the gradle version of Android Studio and the link below was one of the best that I could find.

https://www.geeksforgeeks.org/how-to-update-gradle-in-android-studio/

18518021156, 2021-12-03:
Thank you very much. I successfully completed step 4.

closed 2021-12-03 by 18518021156

MartyG-RealSense, 2021-12-03:
Great news that you succeeded with step 4 - thanks very much for the update!