realsenseai/librealsense #10023: Malfunctioning L515 (malformed vid, device not functioning)

tquintudelft-source, 2021-12-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | Unable to determine (device does not appear) | 
| Driver Version                        | 10.0.19041.1202 |
| Operating System & Version |   Windows 10  | 
| Platform                                 | PC  |
| SDK Version                            |  WIN10-2.49.0.3474                    | 
| Language                            |  Python                          | 
| Segment			|   AR, Holographic content | 

Issue Description: The intel L515 that I am using stopped being identified in the viewer and any other application attempting to use it.

Details:
I have been using an L515 camera for an application I am developing since about around July of this year. I had no problems with it until recently when it has stopped being recognized on my work PC. I originally thought it might be an issue with my PC in particular, but I have since been able to recreate the issue on multiple other PCs. The error is as follows (from the viewer terminal):

``` 
11:35:17 [Info]  - librealsense version: 2.49.0

11:35:17 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2618 - Framebuffer size changed to 2256 x 1410
11:35:17 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2618 - Window size changed to 2256 x 1410
11:35:17 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2618 - Scale Factor is now 1
11:35:18 [Error] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\core\../types.h:310 - hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
11:35:18 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:308 - null pointer passed for argument "device"
11:35:18 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2621 - Couldn't refresh devices - hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
11:35:18 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\win\win-helpers.cpp:147 - malformed vid string: hid_device_system_vhf
11:35:18 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\context.cpp:381 - Found 1 RealSense devices (mask 0xfe) 
```

The camera appears in Device Manager and the troubleshooting diagnosis finds no issues. I am able to access the RGB camera in applications like Zoom or MS Teams, but never the depth camera.

I have attempted to update the firmware of the device with the included firmware update tool and latest firmware image (.bin), but all it does is briefly open the terminal and flash the "malformed vid string" error before exiting. 

Any help would be greatly appreciated :)

MartyG-RealSense, 2021-12-02:
Hi @tquintudelft-source  There was a previous L515 case of the **malformed vid string** message in https://github.com/IntelRealSense/librealsense/issues/7715

Please follow the following sequence of actions.

1.  Unplug the micro-sized end of the USB cable from the base of the camera, turn it around the opposite way (USB-C cables are two-way insertion at the micro-sized end) and re-insert it into the base of the camera.

2.  If this does not resolve the problem, update the RealSense SDK to version 2.50.0 using the installer file in the 'Assets' file list of 2.50.0.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

3.  If this does not resolve the problem, please open the Device Manager and check whether two L515 drivers (Depth and RGB) are listed. 

[image]

If they are listed, right-click on each of these two drivers and select the **Uninstall** menu option to remove it.  When they are both uninstalled, unplug the camera, wait a couple of seconds and then plug the camera back in.  This should cause the two drivers to automatically reinstall.

Do you still experience the problem after trying this uninstall-reinstall, please?

tquintudelft-source, 2021-12-02:
Hi @MartyG-RealSense 

Thanks for your response!

The problem unfortunately still persists after following your recommendations. I have tried the USB-C cable both ways round. 

I'm beginning to think it may be a physical problem with the camera.

MartyG-RealSense, 2021-12-02:
Thanks very much for carrying out those tests.  You have also confirmed that other L515s can work correctly with the same computer.

I also note that you have already attempted a firmware update with a bin file on the affected camera, presumably using the **Update Firmware** option in the RealSense Viewer, and that has failed too.

If swapping the USB cable with one from a working camera does not rectify the problem then I recommend returning the camera to the retailer that you purchased it from using the **Return Material Authorization (RMA)** procedure of that retailer.

MartyG-RealSense, 2021-12-09:
Hi @tquintudelft-source  Do you require further assistance with this case, please?  Thanks!

tquintudelft-source, 2021-12-09:
Hi @MartyG-RealSense, No I think it is okay. I'll try and return it to Amazon. Thanks again for your help!

MartyG-RealSense, 2021-12-09:
Okay, I will keep this case open until you achieve an outcome from your return such as a refund.  Thanks very much for the update!

MartyG-RealSense, 2022-01-22:
Hi @tquintudelft-source   Do you have an update about your camera return that you can provide, please?  Thanks!

MartyG-RealSense, 2022-02-03:
This case will be closed after 7 days from the time of writing this if we do not receive a further comment.  Thanks!

tquintudelft-source, 2022-02-04:
Hi @MartyG-RealSense! Sorry for the inactivity. You can go ahead and close the issue. Thanks again for your help :)

MartyG-RealSense, 2022-02-04:
No problem at all - thanks very much for the update.  As you suggest, I will close the issue.  Thanks again!

closed 2022-02-04 by MartyG-RealSense