realsenseai/librealsense #10025: Kernel Patch not working on 5.4.0-89-generic

yaadhavraajagility, 2021-12-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455} | 
| Firmware Version                   | (05.12.15.50) | 
| Operating System & Version |   {18.04  | 
| Kernel Version (Linux Only)    |  (5.4.0-89-generic)                                         | 
| Platform                                 | Regular Desktop  |

### Issue Description
Kernel Patch doesnt work. I ran the kernel patch succesfully as such:

```
make: Leaving directory '/home/sensorstick/workspace/librealsense/ubuntu-bionic-hwe-5.4'
Patched kernels modules were created successfully

Replacing videodev  -
	Applying the patched module ...  succeeded
Replacing uvcvideo  -
	Applying the patched module ...  succeeded
Replacing hid_sensor_accel_3d  -
Module is resident, unloading ...  succeeded. 
	Applying the patched module ...  succeeded
Replacing hid_sensor_gyro_3d  -
Module is resident, unloading ...  succeeded. 
	Applying the patched module ...  succeeded

Script has completed. Please consult the installation guide for further instruction.
```

Yet, when checking realsense-viewer i didn't see it using the local clock

also `modinfo uvcvideo` shows that there is no realsense there despite being recently compiled

```
 depends:        videobuf2-v4l2,videodev,videobuf2-common,videobuf2-vmalloc,mc
retpoline:      Y
name:           uvcvideo
vermagic:       5.4.0-42-generic SMP mod_unload 
parm:           clock:Video buffers timestamp clock
parm:           hwtimestamps:Use hardware timestamps (uint)
parm:           nodrop:Don't drop incomplete frames (uint)
parm:           quirks:Forced device quirks (uint)
parm:           trace:Trace level bitmask (uint)
parm:           timeout:Streaming control requests timeout (uint)
```

So does the kernel patch even work on 5.4.0?

MartyG-RealSense, 2021-12-02:
Hi @yaadhavraajagility    May I first ask which librealsense SDK version you are using, please?   Official support for kernel 5.4 was listed on SDK version **2.40.0** onwards.

Can you also confirm please that you used the full modinfo instruction `modinfo uvcvideo | grep "version:"` and the string **realsense** was not included in the results?

yaadhavraajagility, 2021-12-03:
2.49.0

```
modinfo uvcvideo | grep "version:"
version:        1.1.1
srcversion:     D6431F013130CC62913C942
```

MartyG-RealSense, 2021-12-03:
Which method did you use to install librealsense, please - with DKMS packages or by building from source code?

With the DKMS package method, the kernel is checked to see if it is updated by using the `modinfo uvcvideo | grep "version:"` instruction after the packages librealsense2-dkms and librealsense2-utils have been installed.  

When building from source code, the kernel patch is applied for Ubuntu 18.04 by running the patch script` ./scripts/patch-realsense-ubuntu-lts.sh`

closed 2021-12-15 by MartyG-RealSense