realsenseai/librealsense #10026: D455 TX2 problem

fwf-lernen, 2021-12-03:
I followed https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md to install librealsense on my TX2.
But when I open realsense-viewer, I can't detect my D455 camera. 
When I run rs-enumerate-devices , I got :"No device detected".
My realsense-viewer version is 2.28.1. Is this the reason why I can't detect D455 camera?

MartyG-RealSense, 2021-12-03:
Hi @fwf-lernen  Support for the D455 camera model in the RealSense SDK was not introduced until **2.38.1** onwards.

fwf-lernen, 2021-12-03:
So how can I install version 2.50.0 on my TX2 board.
Obviuosly, this link :https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md can't do it.

MartyG-RealSense, 2021-12-03:
You could try building 2.50.0 from source code using the Jetson instructions at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

The source code for Jetson can be obtained from the **Source code (zip)** file link in the **Assets** file list at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

fwf-lernen, 2021-12-03:
Must I use Python 3.6 ？

MartyG-RealSense, 2021-12-03:
The RealSense SDK supports up to Python 3.9 at the time of writing this.  Any references to 3.6 in the commands would have to be edited though.

If you do not need to build librealsense's Python compatibility wrapper then you could leave out the Python related commands.

MartyG-RealSense, 2021-12-16:
Case closed due to no further comments received.

closed 2021-12-16 by MartyG-RealSense

fwf-lernen, 2021-12-22:
Thanks a lot, I have successfully solved my problem with your timely and effective help 👍 . Sorry to reply so late : ).

MartyG-RealSense, 2021-12-22:
No problem at all, @fwf-lernen - it's great to hear that you achieved a solution.  Thanks very much for the update!