realsenseai/librealsense #10027: How to align images like rs_camera and convert them to opencv format

zma69650, 2021-12-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   {Win 10   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Win 10  |
| SDK Version                            |  { 2.45 }                          | 
| Language                            |  {c++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I want to use realsense d455 for SLAM in my windows, so I need to get aligned color and depth pictures, just like the ros package provided by realsense. But obviously, I found that the trajectory was off track during the test, so I suspect that there is a problem with my alignment procedure.I want to get the aligned color and depth images and convert them into OpenCV format, just like rs_camera in ROS.My code is as follows:
`
    rs2::pipeline pipe;
    //Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_COLOR, width, height, RS2_FORMAT_BGR8, fps);
    cfg.enable_stream(RS2_STREAM_DEPTH, width, height, RS2_FORMAT_Z16,fps);
    pipe.start(cfg);
    while(1)
    {
        frames = pipe.wait_for_frames();
        // Align to depth 
        rs2::align align_to_depth(RS2_STREAM_COLOR);
        frames = align_to_depth.process(frames);
        //Get each frame
        rs2::frame color_frame = frames.get_color_frame();
        rs2::frame depth_frame = frames.get_depth_frame();

        Mat color(Size(width, height), CV_8UC3, (void*)color_frame.get_data(), Mat::AUTO_STEP);
        Mat pic_depth(Size(width,height), CV_16U, (void*)depth_frame.get_data(), Mat::AUTO_STEP);
   }

`

MartyG-RealSense, 2021-12-04:
Hi @zma69650  The RealSense SDK's OpenCV C++ example program **rs-grabcuts** provides a demonstration of depth-color alignment with OpenCV and **align_to**.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/grabcuts

zma69650, 2021-12-05:
Thanks for your reply @MartyG-RealSense  I did some tests in accordance with the method you provided, but the results are not as expected  [https://github.com/IntelRealSense/realsense-ros/issues/2187](url)

MartyG-RealSense, 2021-12-05:
Thanks very much @zma69650 for the update.  I will look at https://github.com/IntelRealSense/librealsense/issues/url

zma69650, 2021-12-06:
The problem has been solved! ! I'm careless! I wrote the internal reference in depth, it should be in color. thank you for your reply! ! I will close this question.

closed 2021-12-06 by zma69650

MartyG-RealSense, 2021-12-06:
Thanks very much @zma69650 for your updates - it's excellent news that you found a solution  :)