realsenseai/librealsense #10029: Can't install on Windows 11

shanxu9198, 2021-12-05:
When install Intel.RealSense.Viewer.exe  or Depth.Quality.Tool.exe  on Windows 11, crash  message like this.
[image]

MartyG-RealSense, 2021-12-05:
Hi @shanxu9198   Do you experience the same error on Windows 11 if you install the full RealSense SDK on Windows instead of the standalone .exe versions of the Viewer and the Depth Quality tool, please?  The SDK installer will install versions of the Viewer and DQT that are bundled inside the SDK instead of the .exe versions on the Assets file list.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

The SDK installer will place launch shortcut icons for the Viewer and DQT on the Windows desktop when installation has completed.

shanxu9198, 2021-12-05:
@MartyG-RealSense Thank you for your reply. I install Intel.RealSense.SDK-WIN10-2.50.0.3785.exe successfully, but when start  Intel RealSense Viewer I got the same error
[image]
.

MartyG-RealSense, 2021-12-05:
Please try running the SDK example program **rs-capture.exe** that will have been installed on your Windows computer when you installed the full RealSense SDK with Intel.RealSense.SDK-WIN10-2.50.0.3785.exe.   Instructions for launching rs-capture on Windows can be found at https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

MartyG-RealSense, 2021-12-07:
It looks as though **nvhw.inf** is an old Nvidia graphics driver that is associated with the Nvidia Quadro graphics brand.  

Could you provide information about the manufacturer and model name of your computer, please?

Does the error still occur if you update the graphics drivers on your Windows computer?

To help to confirm whether the error is graphics related, could you use the instructions at 
to try running the non-graphical text program **rs-hello-realsense.exe** example program that will be in the same SDK **Tools** folder as rs-capture.exe.

closed 2021-12-19 by MartyG-RealSense