realsenseai/librealsense #10030: can not build realsesne on ubuntu18.04

freepeasantry2021, 2021-12-06:
even the in the readme.md, here has s little description for ubuntu18.04. but when you try to build it,here has so many errors for not installed library and lower library.

do you have some more smart way to build the it on ubuntu18.04 LTS?

MartyG-RealSense, 2021-12-06:
Hi @freepeasantry2021  You could first try building the librealsense SDK with the simple **libuvc backend** installation method in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

freepeasantry2021, 2021-12-06:
@MartyG-RealSense  Than you for your advice. I will try it but i wanto use the v4l2 instead of libuvc

freepeasantry2021, 2021-12-06:
another, when try to patch ubuntu-lts. it always need get  Ubuntu-hwe-5.4-5.4.0-91.102_18.04.1. and will take so long time in china as below
\\\\\\\\\\\\\\\\\\\\\\\\\
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-bionic-hwe-5.4 folder

Package required libelf-dev:  - found
Package required elfutils:  - found
Package required bison:  - found
Package required flex:  - found
Fetching Ubuntu LTS tag Ubuntu-hwe-5.4-5.4.0-91.102_18.04.1  to the local kernel sources folder
remote: Counting objects: 1338479, done.
remote: Compressing objects: 100% (197110/197110), done.
Receiving objects:   1% (13739/1338479), 9.61 MiB | 7.00 KiB/s  

\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\

i firstly get it from https://gitee.com/softmirror/ubuntu-bionic.git into bk-ubunt-bionic/.git
then copy it into the librealsense/ubuntu-bionic-hwe-5.4/.git
but it pops "fatal: You need to specify a tag name"

/////////////
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-bionic-hwe-5.4 folder

Create patches workspace in ubuntu-bionic-hwe-5.4 folder

Package required libelf-dev:  - found
Package required elfutils:  - found
Package required bison:  - found
Package required flex:  - found
Fetching Ubuntu LTS tag   to the local kernel sources folder
fatal: You need to specify a tag name
///////////////////////////////

Does some guy has the similar experience or  know if here has a mirror ?

freepeasantry2021, 2021-12-06:
> I would strongly recommend not using mirror sites to obtain files due to the risk of picking up viruses and malware from the downloads.
> 
> I am not certain from your description which installation method you were using to install librealsense before you tried the mirror. Have you attempted an installation from packages, please?
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

freepeasantry2021, 2021-12-06:
@MartyG-RealSense  i follow this doc:https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

And when run at  "4 Build and apply patched kernel modules for:", it need clone the ubuntu git.

MartyG-RealSense, 2021-12-06:
Unless you need to use CMake to install a RealSense compatibility wrapper such as Python at the same time as building the librealsense SDK from source code then building with packages with the **distribution_linux.md** instructions may be a suitable solution for you.  

Advice provided in https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604 by a RealSense team member about the advantages and disadvantages of libuvc backend versus kernel patching states that patches are wrapped and redistributed with the Debian DKMS package, at least for LTS kernels.

freepeasantry2021, 2021-12-06:
follow https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md,
then run realsense-viewer
conqueror@ubuntu:~/workshop/pkg-librealsense$ realsense-viewer 

conqueror@ubuntu:~/workshop/pkg-librealsense$ modinfo uvcvideo | grep "version"
version:        1.1.2.realsense-1.3.18
srcversion:     51A4A0210E91BE50A7BA2BB
vermagic:       5.4.0-91-generic SMP mod_unload modversions 
conqueror@ubuntu:~/workshop/pkg-librealsense$ 

but here has not depth or color stream in UI.

Would you tell me how to add the local usb source in viewer?

MartyG-RealSense, 2021-12-06:
I commented above in https://github.com/IntelRealSense/librealsense/issues/10030#issuecomment-986547522 about package installation.  I see that you have just performed it.  Thanks very much!

Have you inserted a RealSense camera into the USB port of the computer after launching the Viewer, please?  and if you have, which model of camera is it (D35i, D455, etc)

freepeasantry2021, 2021-12-07:
I use the intel D430.
it works well in win10

I don't know if it impacts the librealsense source code build(cmake without vcpkg) because yestarday i got your advice and built it with packge mode firstly.

And now the librealsense build is finished successfully even thoug i must take a  whole night to udpate the ubuntu-bionic-hwe-5.4.

I don't know if it impacts the librealsense source code build(cmake without vcpkg) because yestarday i got your advice and built it with packge mode firstly.
Thank you very much MartyG-RealSense 

I will figure out why it can not show the color&depth stream

closed 2021-12-07 by freepeasantry2021

MartyG-RealSense, 2021-12-07:
If you have a D430 (the caseless Depth Module Kit board) then it should not be able to stream color as it does not have an RGB sensor.  The **D435** cased model has a built-in RGB sensor.