realsenseai/librealsense #10033: Raspberry pi 4 Install Realsense SDK make -j1 issues

JIeSchnee, 2021-12-06:
I followed the instructions on 
https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435. And adapt the version of protobuf to the latest 3.19.

At the "Install RealSense SDK" make -j1 step. The issues happened:

[ 55%] Linking CXX executable rs-hello-realsense
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to __atomic_load_8' /usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to __atomic_store_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to __atomic_fetch_add_8' collect2: error: ld returned 1 exit status make[2]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:110: examples/hello-realsense/rs-hello-realsense] Error 1 make[1]: *** [CMakeFiles/Makefile2:1240: examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2 make: *** [Makefile:149: all] Error 2

Any solution?

MartyG-RealSense, 2021-12-06:
Hi @JIeSchnee  There is a currently open case at https://github.com/IntelRealSense/librealsense/issues/9962 where a couple of Pi users have the same problem that you have described.  That case has not been resolved at the time of writing this, but reading it should be a useful starting point for investigating your own issue.

MartyG-RealSense, 2021-12-13:
Hi @JIeSchnee  Do you require further assistance with this case, please?  Thanks!

JIeSchnee, 2021-12-15:
Thank you @MartyG-RealSense. I gave up and moved to the ubuntu system. That works well.

MartyG-RealSense, 2021-12-15:
Thanks very much @JIeSchnee for the update.  I'm sorry that Raspbian installation did not work out for you.  As you have moved to Ubuntu, I will close this case.  Thanks again!

closed 2021-12-15 by MartyG-RealSense

sub400, 2022-01-12:
Sorry for posting in this closed thread, but I have exactly the same problem when building on a Raspberry. Unfortunately I cannot change platform.

MartyG-RealSense, 2022-01-12:
Hi @sub400   Does the solution shared by a RealSense user in https://github.com/IntelRealSense/librealsense/issues/9962#issuecomment-998392844 work for you, please?

sub400, 2022-01-12:
Unfortunately it didn't help:

```
[ 89%] Linking CXX executable rs-convert
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_fetch_add_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_load_8'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/convert/CMakeFiles/rs-convert.dir/build.make:134: tools/convert/rs-convert] Error 1
make[1]: *** [CMakeFiles/Makefile2:405: tools/convert/CMakeFiles/rs-convert.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
```

I'm on Raspberry OS Bullseye, just fyi.

MartyG-RealSense, 2022-01-12:
I recall that you previously installed Bullseye on Debian in https://github.com/IntelRealSense/librealsense/issues/9735#issuecomment-983548473 and that went okay.  But you now have problems installing the Raspberry Pi version of Bullseye on a Pi, yes?

There was a Pi case at https://github.com/IntelRealSense/librealsense/issues/9943 that had similar errors to yours.  Their eventual workaround solution - described in https://github.com/IntelRealSense/librealsense/issues/9943#issuecomment-966048785 - was to compile the SDK in the Buster version of the Pi OS and then copy some binary files over to Bullseye.

vjekoslavdiklic, 2022-01-22:
I was able to solve this by using gcc 7 and g++ 7.
```
sudo apt-get install gcc-7 g++-7
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-7 60 --slave /usr/bin/g++ g++ /usr/bin/g++-7
sudo update-alternatives --set gcc "/usr/bin/gcc-7"
```

I used this [post](https://github.com/alexa/avs-device-sdk/issues/1404#issuecomment-531269265) as reference but changed to v7 since v5 and v6 version is not available under Bullseye

MartyG-RealSense, 2022-01-22:
Thanks so much @vjekoslavdiklic for sharing in this discussion what worked for you in your own situation  :)