realsenseai/librealsense #10034: realsense2.dll was not found

ilyak93, 2021-12-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D465 } | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |   {Win 10  | 
                                     | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.49.0. 3474}                          | 
| Language                            |  {C/C++ }                          | 

### Issue Description
I'm trying to run the camera in CLion. The first error was it hasn't recognize the header, but adding the include folder had solved it.

I added to my CMakeLists.txt the cmake command which should be enough to connect it to the project and the cmake doesn't show any errors, but neither it connected the libraries correct.

That is the total look of the CMakeLists.txt file:

```
cmake_minimum_required(VERSION 3.9)

project(data VERSION 0.1 LANGUAGES CXX)

set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

set(WIC_DIR "${CMAKE_CURRENT_LIST_DIR}/../../lib/cmake/WIC/")
list(APPEND CMAKE_MODULE_PATH "${CMAKE_CURRENT_LIST_DIR}/../../lib/cmake/WIC/")

# Save the command line compile commands in the build output
set(CMAKE_EXPORT_COMPILE_COMMANDS 1)

# Make project require C++11
include(CheckCXXCompilerFlag)
CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
CHECK_CXX_COMPILER_FLAG("-std=c++0x" COMPILER_SUPPORTS_CXX0X)
if(COMPILER_SUPPORTS_CXX11)
	set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=c11")
	set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
elseif(COMPILER_SUPPORTS_CXX0X)
	set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
endif()

# Simple non robust way to find the librealsense library
if(WIN32)
	if("${CMAKE_SIZEOF_VOID_P}" STREQUAL "8")
		set(LIBRARY_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\lib\\x64")          # TODO: Update this variable to correct path - folder where realsense2.lib is found
		set(DLL_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\bin\\x64")              # TODO: Update this variable to correct path - folder where realsense2.dll is found
	else()
		set(LIBRARY_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\lib\\x86")          # TODO: Update this variable to correct path - folder where realsense2.lib is found
		set(DLL_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\bin\\x86")              # TODO: Update this variable to correct path - folder where realsense2.dll is found
	endif()
	set(PROJECT_BINARY_OUTPUT_PATH "${PROJECT_BINARY_DIR}/\$\(Configuration\)")                # TODO: Update this variable to correct path - folder to which your project will compile
	set(ADDITIONAL_INCLUDE_DIRS "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\include")   # TODO: Update this variable to correct path - folder where librealsense2 folder is found
endif()

find_library(REALSENSE2_FOUND realsense2 HINTS ${LIBRARY_DIR} REQUIRED)
if(NOT REALSENSE2_FOUND)
	SET(REALSENSE2_FOUND "realsense2")
	message(WARN "Failed to find_library(realsense2)")
endif()

find_package(WIC REQUIRED)
find_package(Pleora REQUIRED)

add_executable(data main.cpp)

target_include_directories(data PUBLIC ${Pleora_INCLUDE_DIR})

target_link_libraries(data PRIVATE WIC::WIC ${Pleora_LIBRARIES} ${REALSENSE2_FOUND})

include_directories(data ${ADDITIONAL_INCLUDE_DIRS})

set_target_properties(data PROPERTIES FOLDER Examples)
```

This is the cmake log:
```
"F:\JetBrains\CLion 2021.2.3\bin\cmake\win\bin\cmake.exe" -DCMAKE_BUILD_TYPE=Debug -DBoost_DIR=F:\WIC_SDK\boost_lib\boost_1_70_0\PREFIX\lib\cmake\Boost-1.70.0 -DCMAKE_DEPENDS_USE_COMPILER=FALSE -G "CodeBlocks - NMake Makefiles" F:\WIC_SDK\examples\wic-module
-- The CXX compiler identification is MSVC 19.29.30137.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: F:/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx86/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Failed
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Failed
-- Found Pleora: C:/Program Files/Pleora Technologies Inc/eBUS SDK/Includes  
-- Found Boost 1.70.0 at F:\WIC_SDK\boost_lib\boost_1_70_0\PREFIX\lib\cmake\Boost-1.70.0
--   Requested configuration: QUIET REQUIRED COMPONENTS container
-- Found boost_headers 1.70.0 at F:/WIC_SDK/boost_lib/boost_1_70_0/PREFIX/lib/cmake/boost_headers-1.70.0
-- Found boost_container 1.70.0 at F:/WIC_SDK/boost_lib/boost_1_70_0/PREFIX/lib/cmake/boost_container-1.70.0
--   libboost_container-vc142-mt-gd-x64-1_70.lib
--   libboost_container-vc142-mt-x64-1_70.lib
-- Adding boost_container dependencies: headers
-- Found Boost: F:/WIC_SDK/boost_lib/boost_1_70_0/PREFIX/lib/cmake/Boost-1.70.0/BoostConfig.cmake (found suitable version "1.70.0", minimum required is "1.70.0") found components: container 
-- Configuring done
-- Generating done
-- Build files have been written to: F:/WIC_SDK/examples/wic-module/cmake-build-debug
```

This is the error that I get when I try to put some code lines which refer to the rs2:
[image]

How can I solve it ?
Thank you in advance.

MartyG-RealSense, 2021-12-06:
Hi @ilyak93   In the discussion https://github.com/IntelRealSense/librealsense/issues/3782 which sounds similar to your situation, a RealSense team member suggested in https://github.com/IntelRealSense/librealsense/issues/3782#issuecomment-486611797 to copy the realsense2.dll file from the RealSense SDK folder location **C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64\realsense2.dll** to the cmake-build-debug folder.

You can find the path that realsense2.dll is located in by installing the full RealSense SDK on Windows.  This can be done by going to the **Assets** file list of a particular RealSense SDK version on the SDK 'Releases' page and then downloading and running the file **Intel.RealSense.SDK-WIN10**, which is an automated installer for the SDK.  

https://github.com/IntelRealSense/librealsense/releases

closed 2021-12-06 by ilyak93

MartyG-RealSense, 2021-12-06:
No problem at all, @ilyak93 - thanks very much for the update.  It's great to hear that you succeeded  :)