realsenseai/librealsense #10035: pyrealsense2 for Mac

Roee1989, 2021-12-07:
I'm trying to install pyrealsense2 with pip and pip3 install on python 3.8 and get the next error:
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
what can I do about it?

MartyG-RealSense, 2021-12-07:
Hi @Roee1989   The pyrealsense2 pip packages are designed for PC computers with x86 / x64 processor architecture.  

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/9687 created a pre-built Mac version of pyrealsense2 with an automated installer that allows use of pip install pyrealsense2 on MacOS, though it was only tested with the Catalina and Big Sur MacOS versions so far.

Roee1989, 2021-12-08:
@MartyG-RealSense Great! It worked for me since I'm using Catalina.
After installation I tried to use the camera with pyrealsense2 and encountered the following error:
 pipeline.start()
RuntimeError: No device connected
I'm using the provided usb cable and when working on PC everything is working.

MartyG-RealSense, 2021-12-08:
Is the case at https://github.com/IntelRealSense/librealsense/issues/10036 created by yourself under a different GitHub account name please, as it seems to describe the same situation.  Thanks!

Roee1989, 2021-12-10:
How do I use the viewer on Mac?
I'm still having the same RuntimeError.
I tried to follow https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md as the thread you linked suggests.
I was able to install and build librealsense as described, I also ran "install_name_tool -change /usr/local/opt/libusb/lib/libusb-1.0.0.dylib @rpath/libusb-1.0.0.dylib librealsense2.dylib" command and copied the files to the framework dir (copied to /Library/Frameworks/Python.framework ) but still no change.
Seems like the installation had no effect.
Any suggestion?

MartyG-RealSense, 2021-12-10:
Are you experiencing the error if you launch the Viewer by inputting **realsense-viewer** into the terminal?

Roee1989, 2021-12-10:
When executing in terminal I get this error:
command not found: realsense-viewer

MartyG-RealSense, 2021-12-10:
Is the Viewer built if you include the CMake term **DBUILD_GRAPHICAL_EXAMPLES=true** in the CMake build instruction to build tools that use graphics such as the Viewer and Depth Quality Tool?

So applying this to the suggested CMake instruction in the official Mac guide:

**cmake .. -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false**

MartyG-RealSense, 2021-12-16:
Hi @Roee1989  Do you require further assistance with this case, please?  Thanks!

Roee1989, 2021-12-17:
I passed the following flags:
cmake -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DBUILD_GRAPHICAL_EXAMPLES=true
Than make -j2 and tried to run the viewer, still same result...
command not found: realsense-viewer

MartyG-RealSense, 2021-12-18:
There was a Mac case at https://github.com/IntelRealSense/librealsense/issues/9319 where the RealSense user in that case also experienced ** command not found: realsense_viewer** on Mac when trying to launch the Viewer.  They eventually found that the realsense-viewer command worked if they built SDK version **2.44.0**

Roee1989, 2021-12-18:
I'm able to launch the viewer manually from folder. In the viewer I can't find a device.

Roee1989, 2021-12-18:
I'm currently using viewer version 2.5. How do I downgrade it to 2.44.0?

MartyG-RealSense, 2021-12-18:
May I first ask which RealSense camera model you are using with the Viewer?

Roee1989, 2021-12-18:
L515

MartyG-RealSense, 2021-12-18:
And is the camera detected in the Viewer if you unplug the micro sized end of the USB cable from the base of the camera, turn the connector around the other way and re-insert it into the camera (USB-C cables are two way insertion at the micro-sized end)

Roee1989, 2021-12-18:
Tried the cable’s micro sized end turn, the viewer doesn’t detect a device.

MartyG-RealSense, 2021-12-18:
I carefully researched how one might approach building an older version of librealsense from source code with CMake on Mac but could not find a clear guide for doing so and I do not have a Mac computer at my location to test with.  I therefore cannot offer advice on doing so with confidence in its correctness, unfortunately.

So to recap, you are using MacOS Catalina, so it is not a problem related to MacOS Monterey.  You also confirmed on a PC computer that your camera hardware works correctly.

Was the PC computer using Windows or Linux?

Can you confirm whether your Mac is an Intel model or a newer Apple Silicon M1 model?

Are you using the camera on a Mac with a USB 3 port?  In a discussion in the link below, a RealSense team member advises that an L515 might encounter problems on a Mac model without USB 3 ports - such as MacBook Pro - that has the camera connected via an adapter to a Thunderbolt port.

https://support.intelrealsense.com/hc/en-us/community/posts/4406814535699-Error-occurs-when-using-L515-on-Macbook

Roee1989, 2021-12-19:
The PC is running on windows 10. The Mac is an intel model with 3.0 USB port.

MartyG-RealSense, 2021-12-19:
Which firmware version is currently installed on your L515, please?  If it is not the current latest version (**1.5.8.1**), could you update the firmware to 1.5.81 on your Windows PC in the RealSense Viewer using the following procedure.

1.  Download firmware version 1.5.8.1 as a 'bin' file from the L515 firmware page.

https://dev.intelrealsense.com/docs/firmware-releases-l500

2.  Launch the Viewer on the Windows PC, go to the 'More' option in the Viewer side-panel and select the **Update Firmware** menu option.  A file selector window will pop up.  Navigate to the downloaded 1.5.8.1 firmware file's location on the computer and select it to commence the firmware update.

[image]

If you update the firmware, does it make a difference to whether the L515 is detected in the Mac version of the Viewer?

MartyG-RealSense, 2022-01-01:
Case closed due to no further comments received.

closed 2022-01-01 by MartyG-RealSense

cansik, 2022-01-11:
@MartyG-RealSense I also experienced the `No device connected` issue on pipeline start with version 2.50.0 of the librealsense library I have prebuilt for python (https://github.com/cansik/pyrealsense2-macosx/issues/2). Getting the current device count works, but starting the pipeline does not. With version 2.49.0, the device can still be opened and accessed. It seems that there is something wrong in the 2.50.0 code base, maybe also related to: https://github.com/IntelRealSense/librealsense/issues/9916#issuecomment-998684578

MartyG-RealSense, 2022-01-12:
Hi @cansik  Thanks again for your excellent pre-built pyrealsense2 project for MacOS!

Thanks too for your report.  Can you add a comment about your 2.50.0 problems on Big Sur to https://github.com/IntelRealSense/librealsense/issues/9916 please so that it can be read by Intel when they investigate the Monterey issues, please?  Thanks!