realsenseai/librealsense #10036: Device not found with pyrealsense2 on Mac

Olimoyo, 2021-12-08:
Hi,

I am using a D435 on a mac OSX Catalina laptop. 

I installed librealsense using the following:
```
brew install librealsense
```
Doing this, I was able to run `realsense-viewer` and get data from the camera. 

I installed pyrealsense2 using `pip install pyrealsense2 -f https://github.com/cansik/pyrealsense2-macosx/releases`. 

I then tried running the following:
```
        import pyrealsense2 as rs
        
        ...

        # Create a pipeline
        pipeline = rs.pipeline()

        # Create a config and configure the pipeline to stream
        #  different resolutions of color and depth streams
        config = rs.config()
        config.enable_stream(rs.stream.depth, stream_width, stream_height, rs.format.z16, stream_fps)
        config.enable_stream(rs.stream.color, stream_width, stream_height, rs.format.bgr8, stream_fps)

        # Start streaming
        profile = pipeline.start(config)
```

and got a `No device found` error. 

Does anyone know how I can get this to work? It works fine on my other laptop running Ubuntu.

MartyG-RealSense, 2021-12-08:
Hi @Olimoyo  May I first confirm whether you have already posted about this problem at https://github.com/IntelRealSense/librealsense/issues/10035 due to the similarity of the case descriptions.  Thanks!

MartyG-RealSense, 2021-12-09:
Let's continue your own case here, separately from https://github.com/IntelRealSense/librealsense/issues/10035

I note first that your program works correctly on your Ubuntu laptop, which indicates that the script is okay.

A Mac user on the Intel Support RealSense forum who was using Big Sur also experienced the same **RuntimeError: No device connected** error recently when starting the pipeline (see the link below).  On the same discussion, another RealSense user who was using Monterey also reported experiencing the error. 

https://support.intelrealsense.com/hc/en-us/community/posts/4411636626195--No-device-connected-error-on-mac

My interpretation of the Big Sur user's description of their case in the link below is that the *RuntimeError: No device connected** error was resolved when they repeated installation of librealsense again.

https://support.intelrealsense.com/hc/en-us/community/posts/4411636626195/comments/4412047635731

Olimoyo, 2021-12-10:
I tried re-installing it in a few different ways and nothing seemed to have worked.

I ended up changing my setup to avoid using my Mac so I'll close this issue, but if anyone finds a solution I'd be curious to know!

closed 2021-12-10 by Olimoyo

MartyG-RealSense, 2021-12-11:
Thanks very much @Olimoyo  for the update.  I'm sorry that the hoped-for outcome was not achieved in this case.