realsenseai/librealsense #10038: Set font size / font scale for realsense-viewer

hg-zt, 2021-12-08:
Hi, I found characters in `realsense-viewer` are too small for my monitor, how can I adjust the font?

MartyG-RealSense, 2021-12-08:
Hi @hg-zt   The RealSense Viewer does not have a setting to adjust the font size.  You could try pressing the **F8** keyboard key to toggle Fullscreen mode on and off so that the Viewer is stretched to the full size of your display to see whether that helps to enlarge the text.

hg-zt, 2021-12-08:
Thanks. I've tried to use full screen mode but the font size is too small for the DPI. And I'm using linux which unfortunately do not have a unified DPI scaling solution.

It seems to be related to https://github.com/ocornut/imgui/issues/1676.

closed 2021-12-08 by hg-zt

Peixuan-Wu, 2024-03-12:
@hg-zt  Hi, did you resolve this problem?