realsenseai/librealsense #10039: Unable to receive frames from external software trigger on a FRAMOS D435e sensor

ashwinmudigonda2, 2021-12-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Framos D435e  | 
| Firmware Version                   | 2.33.10 | 
| Operating System & Version |   Ubuntu 17  | 
| Kernel Version (Linux Only)    |  4.15.0-142-generic                                       | 
| Platform                                 | PC  |
| SDK Version                            |  2.33.10                          | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description
I am unable to read frames when the RealSense device is set to External camera sync mode and I send a software trigger. I am using a FRAMOS D435e sensor.

I create the realsense object as usual.
```
sensors = self._pipeline.get_active_profile().get_device().query_sensors()
for sensor in sensors:
   sensor.set_option(rs.option.ext_trigger_source, 2) # Software trigger
   if sensor.is_depth_sensor():
      sensor.set_option(rs.option.inter_cam_sync_mode, 3)
      sensor.set_option(rs.option.software_trigger_all_sensors, 1)
   else:
      sensor.set_option(rs.option.inter_cam_sync_mode, 1)
```

Later, I try to read a color & depth frame by:

```
device = self._pipeline.get_active_profile().get_device()
ds = device.first_depth_sensor()
ds.set_option(rs.option.software_trigger, 1) # From the rs-software-trigger.cpp example, the depth sensor's trigger fetches RGB and depth frames
```

And I fetch the frames by:

```
frames = self._pipeline.poll_for_frames()
```

But when I try to get any info from any of the frames, I get an error:
```
ERROR    null pointer passed for argument "frame"
```

And the condition:

```
if not frames:
   continue
```

always holds true. 

The C++ example that came shipped with the library (examples/software-trigger) works fine and I tried to follow the sequence of steps from it in my code. But I cannot seem to receive valid frames. What am I doing wrong?

MartyG-RealSense, 2021-12-09:
Hi @ashwinmudigonda2   Intel do not provide technical support for the FRAMOS range of RealSense-compatible 'e' cameras as they are not Intel products.  Please contact FRAMOS support with your question by visiting their support page for their 'e' camera models at the link below and clicking the 'Contact Us' or 'Let's Talk' buttons on the page.

https://www.framos.com/en/industrial-depth-cameras#downloads  

FRAMOS also provide a PDF document at the link below on the subject of synchronizing the camera to an external event.

https://www.framos.com/framos3d/D400e/Documentation/Application%20Notes/Framos_D400e_External_Event_Camera_Synchronization_AppNote.pdf

ashwinmudigonda2, 2021-12-12:
Thank you. I am able to replicate the example and get it to work in C++. It just doesn't work with the Python bindings. I will contact Framos.

closed 2021-12-12 by ashwinmudigonda2