realsenseai/librealsense #10040: depth frame error: large part of depth info missing

surefyyq, 2021-12-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |    Linux (Ubuntu 18.06) | 
| Platform                                 | NVIDIA Jetson |
| Language                            |  python   | 

In an operation environment, the realsense camera has been working normally, but out of nowhere the following phenomenon appeared:
[image]
[image]

It seems one of the imager fail to function at that time.
Alignment is done and re-plug camera solved the problem, but why is that and how to avoid？

MartyG-RealSense, 2021-12-09:
Hi @surefyyq  As the camera is pointing down towards the soil, it does not look like an infrared disruption event that is triggered by an infrared sensor on the camera being exposed to a directly observed flash of light such as sunlight.

There are also infrared disruption events that can occur without the presence of a strong light source, such as in a windowless indoor room.

This does not look like a typical infrared corruption though, where the depth image may be very low-detail and broken looking as a result.  The neatness of the depth image makes it look as though the entire image from one of the left and right infrared sensors became unavailable until the camera was reset (the 400 Series stereo depth cameras generate the depth frame from the raw left and right infrared frames that are captured).

I achieved a similar image to your one by placing a thick coffee-cup coaster over the left infrared sensor of a D435i camera model to block it, though the presence of some noise on the obscured side of my image compared to the total emptiness on yours suggests that on your camera, the left infrared imager was not capturing at all at that time.  

[image]

I note though the orange band on the left side of your image whose position roughly corresponds to the distance that the light grey border on the RGB image is from the eggs.  This gives the impression that the entire depth image is being pushed rightwards, otherwise more of the area of empty soil to the right of the eggs would likely be shown on the depth image.  

If you are performing depth-color alignment (as you mentioned that "alignment is done") then there might be a failure in the alignment operation in your program.  Since the problem can be corrected by a camera reset, this makes it more likely that it is a software issue rather than a camera hardware fault.  

Whatever the cause is, if a camera reset corrects it then a simple solution may be to add a **hardware_reset()** instruction to your Python script with a timer that automatically performs the reset every 'x' number of minutes or hours.

surefyyq, 2021-12-13:
Hi@MartyG-RealSense , thanks for your reply.
I encountered this issue for the first time since I used the camera 1 year ago, so I guess it is a very low probability failure, I will add hardware_reset() and continue to observe.

MartyG-RealSense, 2021-12-13:
Thanks very much, @surefyyq - I look forward to your next update.  Good luck!

MartyG-RealSense, 2021-12-25:
Hi @surefyyq  Do you have an update about this case that you can provide, please?  Thanks!

surefyyq, 2021-12-27:
Hi @MartyG-RealSense 
Due to the complexity of my pre-and-post process function, adding hardware_reset() is not progrossing well. I'm keeping it running until the problem reappears, before that I should close this issue.

closed 2021-12-27 by surefyyq

MartyG-RealSense, 2021-12-27:
Okay, thanks very much @surefyyq for the update.  Good luck!