realsenseai/librealsense #10041: Failed to build Could NOT find Threads (missing: Threads_FOUND)

Obrepal, 2021-12-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |

| Operating System & Version |   {MacOS} Monterey 12.0.1   | 
                                     

### Issue Description
<Describe your issue / question / feature request / etc..>

I do have much experience with CMake. 
I downloaded folder using: 
`it clone https://github.com/IntelRealSense/librealsense`

and then followed the steps for MacOS 

When I want to build it 
`cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
`

This occurs 
`build % cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -- Checking internet connection... -- Internet connection identified -- Info: REALSENSE_VERSION_STRING=2.50.0 -- Setting Unix configurations -- using RS2_USE_LIBUVC_BACKEND CMake Error at /opt/homebrew/Cellar/cmake/3.22.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message): Could NOT find Threads (missing: Threads_FOUND) Call Stack (most recent call first): /opt/homebrew/Cellar/cmake/3.22.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE) /opt/homebrew/Cellar/cmake/3.22.0/share/cmake/Modules/FindThreads.cmake:238 (FIND_PACKAGE_HANDLE_STANDARD_ARGS) examples/measure/CMakeLists.txt:9 (find_package)`

I don't want to make it too long but here is  CMakeError.log

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /Library/Developer/CommandLineTools/usr/bin/c++
Build flags:
Id flags:

The output was:
1
ld: library not found for -lc++
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /Library/Developer/CommandLineTools/usr/bin/cc
Build flags:
Id flags:

The output was:
1
ld: library not found for -lSystem
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Determining if the include file pthread.h exists failed with the following output:
Change Dir: /Users/Obrepal/librealsense/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make -f Makefile cmTC_5e8f1/fast && /Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/cmTC_5e8f1.dir/build.make CMakeFiles/cmTC_5e8f1.dir/build
Building C object CMakeFiles/cmTC_5e8f1.dir/CheckIncludeFile.c.o
/Library/Developer/CommandLineTools/usr/bin/cc -pedantic -g -D_DEFAULT_SOURCE -mfpu=neon -mfloat-abi=hard -ftree-vectorize -latomic -pthread -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.0.sdk -fPIE -MD -MT CMakeFiles/cmTC_5e8f1.dir/CheckIncludeFile.c.o -MF CMakeFiles/cmTC_5e8f1.dir/CheckIncludeFile.c.o.d -o CMakeFiles/cmTC_5e8f1.dir/CheckIncludeFile.c.o -c /Users/Obrepal/librealsense/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
clang: warning: -latomic: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfpu=neon' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfloat-abi=hard' [-Wunused-command-line-argument]
Linking C executable cmTC_5e8f1
/opt/homebrew/Cellar/cmake/3.22.0/bin/cmake -E cmake_link_script CMakeFiles/cmTC_5e8f1.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/cc -pedantic -g -D_DEFAULT_SOURCE -mfpu=neon -mfloat-abi=hard -ftree-vectorize -latomic -pthread -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.0.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/cmTC_5e8f1.dir/CheckIncludeFile.c.o -o cmTC_5e8f1
ld: library not found for -latomic
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [cmTC_5e8f1] Error 1
make: *** [cmTC_5e8f1/fast] Error 2

MartyG-RealSense, 2021-12-09:
Thanks very much @Obrepal for the highly detailed information.  It is a good rule of debugging to start with the first error that occurs in a log, as subsequent errors may be occurring as a consequence of that original problem.  As you highlighted, the first error that is listed is **Could NOT find Threads**.

A RealSense M1 user who encountered this same error shared a workaround at https://github.com/IntelRealSense/librealsense/issues/8250#issuecomment-768309524 that enabled CMake to build correctly.  Does the procedure at this link resolve this error for you too, please?

Obrepal, 2021-12-09:
so I added it to CMakeLists.txt 
but still something is not right 
```
CMake Error at examples/measure/CMakeLists.txt:12 (add_executable):
  Target "rs-measure" links to target "Threads::Threads" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at tools/convert/CMakeLists.txt:17 (add_executable):
  Target "rs-convert" links to target "Threads::Threads" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
```

MartyG-RealSense, 2021-12-09:
There is a librealsense example of this 'Threads' CMake error at https://github.com/IntelRealSense/librealsense/issues/5656

In a non-librealsense discussion, someone who encountered the same issue with CMake shared the following advice.

*********

I solve this problem by adding **find_package(Threads)** to the CMakeLists.txt file in the root directory.

Obrepal, 2021-12-09:
ok, so now this `ld: library not found for -latomic` seems to be last problem (at least in this part 👯 )

MartyG-RealSense, 2021-12-10:
Thanks very much for your patience.  There is advice about **ld: library not found for -latomic** at https://github.com/IntelRealSense/librealsense/issues/9999#issuecomment-979891819 and the comment beneath it which advises using XCode to build librealsense  instead of 'make -j'.

Obrepal, 2021-12-10:
Come on, without you I would not know what to do. Thanks a lot I really appreciate it.
Ok, so I delated CMakeCache from previous build. 

Used this alternative formula from mentioned comment. 
And now I think I am ready to go. (?)
`-- Build files have been written to: /Users/Obrepal/.../.../librealsense/build
`
Or there is something else worth doing?

MartyG-RealSense, 2021-12-10:
Thanks very much for your kind words.  :)  I do not think that you are doing anything wrong, as building librealsense for M1 Macs does tend to be a step by step process of error-correcting for most M1 users at present.

When the pyrealsense2 wrapper has been successfully built on Mac, you should be able to locate the files 'pyrealsense2.so' and 'pybackend2.so' in the folder location **build > wrappers > python > Debug**

Obrepal, 2021-12-10:
I am really starting to wish it was Linux. M1 is like being stabbed on every step.
Unfortunately, my lack of knowledge doesn't let me build it in XCODE as here https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk. Should I open new project, select build folder (/librealsense/build) and then simply select Product -> Build in XCode?

MartyG-RealSense, 2021-12-10:
I apologize for the difficulties that you are experiencing.  I do not have a Mac computer to test with at my location, unfortunately.

https://github.com/IntelRealSense/librealsense/issues/1671 has advice about problems with building with XCode on Mac.

Obrepal, 2021-12-10:
There is no need to apologize, I just have to learn more, as I mentioned before  building stuff from Github is something new for me and also M1 doesn't want to cooperate so it is hard to start. I just hope that I will manage to build it and won't give up.

MartyG-RealSense, 2021-12-10:
I researched your case from the beginning in search of fresh insights.  Most RealSense users attempting to install on an M1 computer likely start with a common reference source, such as the GitHub version of the official Mac installation guide, which was last updated in September 2021 at the time of writing this and so has advice that is relatively up to date.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

Once the guide is followed and an error occurs on M1, this is the point where users tend to have a different installation experience.  For example, they may experience similar errors but not in the same order as other users, and so they refer to the M1 installation issues on the forum to find pieces of information for the particular error that they currently have.  This makes it currently difficult to assemble a single universal instruction guide that will work for a majority of M1 users.

Obrepal, 2021-12-10:
I am aware of it.
Tomorrow I will try to reinstall it, now I know how to solve some issues
and maybe other were created by mistake (while solving previous ones)  so
it might help a little.

MartyG-RealSense, 2021-12-16:
Hi @Obrepal  Do you have an update that you can provide about this case, please?  Thanks!

Obrepal, 2021-12-16:
Not yet, I had to take a rest.
I will try again next week.  Maybe this time will be easier. I will let you
know in case of any problem

MartyG-RealSense, 2021-12-16:
Okay, thanks very much for the update - good luck!

MartyG-RealSense, 2021-12-28:
Hi @Obrepal  Do you have a further update about this case that you can provide, please?  Thanks!

MartyG-RealSense, 2022-01-02:
Case closed due to no further comments received.

closed 2022-01-02 by MartyG-RealSense

Obrepal, 2022-02-22:
Sorry, my other projects kept me busy.
So I made cmake and it seems fine.
`cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false`

 ```
Configuring done
-- Generating done
-- Build files have been written to: /Users/Obrepal/Desktop/Test_real_sense/librealsense/build
```

But then I run into this:
` make -j2

[  2%] Built target fw
[  2%] Performing update step for 'libusb'
[  4%] No patch step for 'libusb'
[  9%] Built target realsense-file
[  9%] Performing configure step for 'libusb'
[  9%] Linking CXX executable rs-embed
CMake Deprecation Warning at CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

ld: library not found for -latomic
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [tools/embed/rs-embed] Error 1
make[1]: *** [tools/embed/CMakeFiles/rs-embed.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/Obrepal/Desktop/Test_real_sense/librealsense/build/libusb-prefix/src/libusb-build
[  9%] Performing build step for 'libusb'
Consolidate compiler generated dependencies of target usb
[100%] Built target usb
[  9%] Performing install step for 'libusb'
[100%] Built target usb
Install the project...
-- Install configuration: ""
-- Installing: /Users/Obrepal/Desktop/Test_real_sense/librealsense/build/libusb_install/lib/libusb.a
[  9%] No test step for 'libusb'
[  9%] Completed 'libusb'
[ 10%] Built target libusb
make: *** [all] Error 2`

I really feel helpless but will check this one more time.
Is there chance to make this more M1 friendly ? 😄

MartyG-RealSense, 2022-02-22:
I note that your build log has a deprecation warning about the CMake version.

**CMake Deprecation Warning at CMakeLists.txt:4 (cmake_minimum_required): Compatibility with CMake < 2.8.12 will be removed from a future version of CMake.**

The message would suggest that a CMake version less than 2.8.12 is being detected.  Intel's official MacOS installation instructions state that CMake 3.8 or newer is required by librealsense.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md#building-from-source

Obrepal, 2022-02-22:
According to brew I have: 
 brew list --versions cmake
**cmake 3.22.2**

MartyG-RealSense, 2022-02-22:
It is worth noting that there have been problems with a number of RealSense users in recent times with using RealSense on Monterey, as described in https://github.com/IntelRealSense/librealsense/issues/9916 - this has affected Monterey using Macs with both Intel and M1 processors.

Obrepal, 2022-02-22:
Ok, so we are waitng for patch from Intel?

MartyG-RealSense, 2022-02-22:
Intel concluded that the problem needs to be addressed from Apple's side in MacOS rather than implementing a fix in the RealSense SDK.

A RealSense M1 user did manage to build on Monterey at https://github.com/IntelRealSense/librealsense/issues/9916#issuecomment-1026756545 - though with errors and another RealSense user at https://github.com/IntelRealSense/librealsense/issues/9916#issuecomment-1048069662 speculated that it may be related to camera permissions on MacOS.

Obrepal, 2022-02-22:
> A RealSense M1 user did manage to build on Monterey at #9916 (comment)
> - though with errors and another RealSense user at #9916 (comment)

Ok, so we are stuck again.
Thanks a lot for help

mattbeghin, 2022-04-29:
So I could get my app to build with RealSense on Apple M1 / Monterey, but I can't ask my users to start my app with a sudo...
Let me know if I'm correct: Monterey doesn't allow controlling power state of a device through libusb without being root, is that correct ?
I tried getting more rights with AuthorizationCopyRights, forcing user to enter admin password, but it doesn't help. I suppose the only way actually would be to start an external process with sudo and use shared memory, but I suppose I'll just wait for apple to address this issue...

MartyG-RealSense, 2022-04-29:
Hi @mattbeghin  There has been significant progress with Mac installation on Monterey and on M1 Macs recently, as described at  https://github.com/IntelRealSense/librealsense/issues/9916#issuecomment-1107776319