realsenseai/librealsense #10045: Why does it take numerous attempts to do On-Chip Calibration?

ravijo, 2021-12-10:
Dear RealSense Team,

I am using Intel RealSense Viewer v2.44.0 application to do On-Chip Calibration (OCC) on a D415 camera.

-----------------------------------------------------------
| Required Info               |                           |
|-----------------------------|---------------------------|
| Camera Model                | D415                      | 
| Firmware Version            | 05.12.12.100              | 
| Operating System & Version  | Ubuntu 18.04.6 LTS        | 
| Kernel Version (Linux Only) | 5.4.0-86-generic          | 
| Platform                    | PC                        |
| SDK Version                 | 2.44.0-0~realsense0.4389  | 
| Language                    | None                      | 
| Segment			          | Robot                     | 

### Issue Description
Even after doing OCC, it is not saving the values correctly. Because the next time I did OCC, it showed "calibration required" again. I applied the calibration and again did OCC. It showed "calibration required" even the 3rd time. Then I stopped the realsense-viewer and removed the USB. I plugged it again, tried OCC, and it showed "calibration required".

Please see following screenshots showing  numerous attempts to do On-Chip Calibration:

1. Attempt 1

[image]

2. Attempt 2

[image]

3. Attempt 3

[image]

4. Attempt 4

[image]

What is happening here? Am I missing something?

MartyG-RealSense, 2021-12-10:
Hi @ravijo  In your other On-Chip Calibration case at https://github.com/IntelRealSense/librealsense/issues/10044 you mentioned that you had tried another D415 camera.  Do you receive the same Requires Calibration message with both cameras, please?

ravijo, 2021-12-11:
Hello @MartyG-RealSense 

I encountered that performing OCC successfully is very tricky and intermittent. It took me a couple of hours after repeatedly restarting the PC and the realsense-viewer. I checked it on the other D415, but no success. Unfortunately, I was running out of time due to this exercise, gave up, and started looking for expert advice from you.

MartyG-RealSense, 2021-12-11:
Intel's white-paper guide document for On-Chip calibration provides a scale that indicates how the tool decides to grade the health check score.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

[image]

This image has an accompanying note in the paper that states: "The “Health-Check” indicator will return a measure of the need for recalibration.  While units are operational for values below 1, it is optimal to have an absolute value below 0.25.  This health-check does not need any special target".

So if a health check score significantly deviates from what is considered a 'good' score then the **Requires Calibration** recommendation may be displayed.

If you are provided with a score graded with 'Requires Calibration' then I would recommend repeating the calibration until you get a better grade such as 'Could Be Improved' or 'Good' (if you are able to achieve Good) and then write that to the camera with the **Apply New** button.

ravijo, 2021-12-11:
Hello @MartyG-RealSense 

Thank you for the quick response.

> If you are provided with a score graded with 'Requires Calibration' then I would recommend dismissing the calibration and repeating the process until you get a better grade such as 'Could Be Improved' or 'Good' (if you are able to achieve Good) and then write that to the camera with the Apply New button.

Does it mean I need to keep repeating the calibration until I get "Good"? If yes, typically, how many attempts does it take? Further, what do you think about the screenshots of 4 attempts shared in the beginning of this issue.

MartyG-RealSense, 2021-12-11:
Calibration can be repeated as many times as you wish until you get a result that is satisfactory to you.  Once that happens, you can write the calibration to the camera hardware.  You do not need to write it until you are satisfied.  You could then perform further calibrations to try to improve the result, and only save the calibration when you achieve a better result than the previously stored one.

As mentioned in your other case, you can also perform a 'gold reset' to restore the original factory-new settings of the camera.

In regard to the images in the original message, they are good quality in terms of being solid with few holes but they should ideally be mostly a single color right across their surface instead of showing a spectrum of different colors.  This is because the depth values are color-shaded from blue (near range to the camera) to red (far range).  So the color should progressively shift over distance from blue in the foreground close to the camera to red at far range.  Because the wall is at relatively the same difference from the camera, it should be almost the same color right across it.

A spectrum of colors across the wall surface suggests that the camera is having difficulty accurately analyzing the wall for depth detail.  This could be because the wall has low texture or no texture, or it is painted a dark color such as dark grey or black that a depth camera will have difficulty reading (because it is a general physics principle that such color shades absorb light).

***********

You can also edit camera calibration information by going to the **More** option in the Viewer and selecting the **Calibration Data** option from its menu.

[image]

In this interface, left-clicking on the box beside the option **I know what I'm doing** in the bottom corner unlocks a button called **Restore Factory** that restores the calibration in the camera hardware's flash storage to factory settings.  You may see a blue flash when clicking this button.  This  is normal and indicates that the displayed calibration values have updated to the factory defaults.

[image]

ravijo, 2021-12-12:
Hi @MartyG-RealSense 

> Calibration can be repeated as many times as you wish until you get a result that is satisfactory to you. Once that happens, you can write the calibration to the camera hardware. You do not need to [...]

In my case, I did OCC 4 times and every time applied new calibration and tried it again. The 4 screenshots posted above shows "Health Check" values. You can see how it is changing over the attempts and saying "Requires Calibration" even after applying new calibration every time.

So my question is "why is it taking so many attempts to perform OCC?"

> A spectrum of colors across the wall surface suggests that the camera is having difficulty accurately analyzing the wall for depth detail. This could be because the wall has low texture or no [...]

The camera was facing towards white color wall at a distance of 60 cm approximately.

MartyG-RealSense, 2021-12-12:
I ran another series of On-Chip calibration tests with a D415.  I found that the best results were consistently achieved when using the White Wall setting.

I managed to achieve an image somewhat resembling your 'color spectrum' images if I directed the camera at an area with a strong light source.  Do you have a strong light projecting onto the surface of the plain wall that you are performing the calibration on?

MartyG-RealSense, 2021-12-18:
Hi @ravijo  Do you require further assistance with this case, please?  Thanks!

ravijo, 2021-12-18:
@MartyG-RealSense 

Unfortunately, I still have little confidance to perform On-Chip Calibration. For example, can you tell me what type of wall do you require for OCC? What color, texture, distance, surface, etc are best suited for OCC.

MartyG-RealSense, 2021-12-18:
Section 2.4 of the guide for On-Chip calibration provides a graphic illustrating the kinds of textures that have been tested successfully.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-2-4-extending-to-beyond-simple-flat-textured-target

ravijo, 2021-12-18:
Thank you very much.

It seems that we don't need a flat surface. It works well with any surface. (Image D on Figure 6)

Can you please confirm my understanding before closing this issue?

MartyG-RealSense, 2021-12-18:
Having a textured surface is helpful for depth analysis in general.  For untextured or low textured surfaces, having the dot pattern projection enabled is recommended.  

Figure 6 highlights how the pattern projection is turned off in outdoor scenes in bright sunlight.  This is because the 400 Series cameras can make use of ambient light in a scene for analyzing surfaces for depth information, so the assistance of the dot pattern (which can be used by the camera for depth analysis in the absence of strong light) may not be required.

ravijo, 2021-12-24:
@MartyG-RealSense 

I gave up doing OCC!

MartyG-RealSense, 2021-12-24:
Okay, thanks very much @ravijo for the update.  I'm sorry that you didn't achieve the desired results in your particular case.

closed 2021-12-30 by MartyG-RealSense