realsenseai/librealsense #10046: No Depth Frame Received

mmmarcopalma, 2021-12-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515  | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC |
| SDK Version                           | 2.45.0 | 
| Language                               | Python | 
| Segment			        |  Robot | 

### Issue Description
The sensor fails to stream the Depth Frame, both from the RealSense Viewer and from Python 3.9.5 code. 
Since my application requires the sensor being 5m away from the PC, I have been using a USB 3.2 cable extension for more than 3 weeks without problems. 

Here the details: 
The RealSense viewer recognizes the device as USB 3.2
The RGB Camera Module and the Motion Module are perfectly streaming.  

This a screen capture of the Viewer when opening it for the first time after plugging the Camera to USB and enabling the Depth Sensor:

[image]

And this is a screen capture of the Viewer after disabling and re-enabling the Depth Sensor:

[image]

If I try to stream a frame from code, the Python terminal outputs the following exception:

`hr returned: HResult 0x8007001f: "A device attached to the system is not functioning.`

After the issue happened for the first time, I tested several options suggested in other similar issues, with no luck:
1. Flipping USB C plug orientation.
2. Performing Hardware reset from the Viewer.
2. Updating Firmware from the Viewer.
3. Updating SDK to v2.48.0, v2.49.0 and v2.50.0 ( with v2.49 and v2.50 the device is not even recognized by the Viewer). 
4. Uninstalling the device from Windows Device Manager to trigger an automatic driver update.
5. Testing the depth module at the lowest resolution. 
6. Testing all USB ports on my machine. 

Here the info from the Viewer:
[image]

Thank you for your help.

MartyG-RealSense, 2021-12-11:
Hi @mmmarcopalma   Does depth streaming work normally if you attach the camera directly to the computer using the official 1 meter cable supplied with the camera without an extension cable?  And can you also confirm please that you have no other RealSense-using programs already running when you launch the Viewer or your Python program?

mmmarcopalma, 2021-12-11:
Hi @MartyG-RealSense, thanks for the prompt feedback. 
I can confirm both of your questions, I have no other active RealSense processes and the exception raised by the Viewe and Python are the same with the original cable.

mmmarcopalma, 2021-12-11:
Maybe you can also report this strange behavior.
After updating to SDK v2.49 or v2.50 and to the latest firmware version,  the Camera is not detected by the Viewer (even if Windows Device Manager does) and a null pointer exception is raised:

[image]

MartyG-RealSense, 2021-12-11:
Do you experience the same errors if you download and run the standalone **Intel.RealSense.Viewer.exe** version of he Viewer from the 'Assets' file list on the SDK 'Releases' page?

https://github.com/IntelRealSense/librealsense/releases

[image]

The .exe file should be launched with a double-leftclick from the folder location that it was downloaded to and not with the launch shortcut icon on the Windows desktop.

mmmarcopalma, 2021-12-11:
after executing the standalone version as you described, it seems two devices are found, but neither of those is detected nor can be enabled:

[image]

MartyG-RealSense, 2021-12-11:
Thank you very much.  Next, please check the Windows privacy settings using the instructions in the link below to see whether access to the camera hardware and permission for applications to access the camera are both set to On.

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

mmmarcopalma, 2021-12-11:
Both access to hardware and application permissions were already set to On.

MartyG-RealSense, 2021-12-11:
Thanks for your continued patience during this diagnostic process.  Next, please check whether depth and RGB can be streamed by the **rs-capture** example program using the instructions at https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

mmmarcopalma, 2021-12-11:
Thank you for the support. 
After execution, both the terminal and the application windows stand idle for about 30 seconds and they suddenly close without outputting any information. Seems like a timeout occurred.

MartyG-RealSense, 2021-12-11:
Taking all of the above test information in this case into consideration, my final conclusion is to recommend that the camera is returned to the retailer that it was purchased from using the **Return Material Authorization (RMA)** returns procedure of that retailer.

mmmarcopalma, 2021-12-11:
Thank you, I hope the process will go through.

mmmarcopalma, 2021-12-12:
Hi @MartyG-RealSense, the retailer will not accept a return since the invoice is older than 30 days. 
 Shouldn't I be able to send it back to Intel and ask for a Warranty Exchange?

MartyG-RealSense, 2021-12-12:
I will seek advice from Intel about your case.  It will likely be tomorrow when I receive a response.

In the meantime, may I ask whether your camera had been stored inactive in a cool dry place such as a cupboard or drawer for a period of weeks / months before you discovered the depth problem?

mmmarcopalma, 2021-12-12:
The camera has been (successfully) tested immediately after arrival and used for about a week.
It has then been stored inactive for some months in its packaging inside a box in a dry place.  
3 weeks ago it was re-activated and used for about 3 weeks until the depth problem raised. 

Thanks for the help, I'll wait for some news tomorrow. 

In the meanwhile, could you please tell me when I can find the StockingID/Part# which is required in the online warranty check? Thank you

MartyG-RealSense, 2021-12-12:
Depth stream failure after a prolonged period of inactive storage is a known issue that has been reported in a small number of cases.  So far it has only been reported on the 400 Series stereo depth models rather than L515 lidar.  In the 400 Series cases, the problem resolved itself a few weeks after coming out of storage.  For example, one camera was returned to California from India and was found to be working upon receipt after the international shipping.  So there may be environmental / temperature factors involved in the recovery.

Typically, that depth problem manifests during first use after storage rather than a few weeks later on the 400 Series cameras, though the L515 technology is of course different.

I suggested to one customer with a non-working depth stream after storage to try applying power to the camera once a day for a few weeks to see whether it restored the camera to working order.

The product / part ID code for the L515 model is **999NGF**

mmmarcopalma, 2021-12-13:
Hi @MartyG-RealSense, do you have any news regarding the warranty issue? I bought the item in July 2021, less tha 1 year.  I would like to start this process as soon as possible since I need the camera to be working and it seems likely that there is a hardware malfunctioning. 
Thank you,
Marco

MartyG-RealSense, 2021-12-13:
Hi @mmmarcopalma   I do not have information yet regarding your warranty query.   I will update here as soon as I receive feedback from Intel.  Thanks again for your patience.

MartyG-RealSense, 2021-12-14:
Hi again @mmmarcopalma  Intel have asked if you can please provide a postal shipping address for your location.  You can email the address to me at:

martyx.grover@intel.com

mmmarcopalma, 2021-12-14:
Thank you, I've just reached you by email.

MartyG-RealSense, 2021-12-14:
Thanks very much.  I have not received the email yet.  You could alternatively post the address in a comment and then delete the comment as soon as I have received it.

mmmarcopalma, 2021-12-14:
Hi @MartyG-RealSense, 
maybe it ended up in the spam folder for some reasons. 
Can you try to reach me at this address please?
marcopalma.posta@gmail.com

Thank you,
Marco

MartyG-RealSense, 2021-12-14:
I checked the spam folder but it was not there.  I will send you an email that you can reply to.

mmmarcopalma, 2021-12-14:
I've answered you, did you receive?

MartyG-RealSense, 2021-12-14:
I did receive and reply - thanks very much.

MartyG-RealSense, 2021-12-14:
By the way, feel free to edit out your email address at https://github.com/IntelRealSense/librealsense/issues/10046#issuecomment-993486781 if you do not want it to be shown publicly.

MartyG-RealSense, 2022-01-22:
Hi @mmmarcopalma  Have you now received a replacement camera, please?  Thanks!

MartyG-RealSense, 2022-02-03:
This case will be closed after 7 days from the time of writing this if we do not receive a further comment.  Thanks!

closed 2022-02-09 by MartyG-RealSense