realsenseai/librealsense #10048: no frame received

hayrettin46, 2021-12-12:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|112322076662|
|**Firmware Version**|05.13.00.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|108223052418|
|**Firmware Update Id**|108223052418|

Please provide a description of the problem

MartyG-RealSense, 2021-12-12:
Hi @hayrettin46  Could you provide more information about your problem please, including any warning / error messages.  

Which stream type (RGB, Depth or Motion Module) is the No Frames Received message displayed on?

Also, are you enabling all three stream types (RGB, Depth or Motion Module) at the same time?

hayrettin46, 2021-12-12:
Hi

I installed sdk on 2 different laptops, in one of them the pyrealsense2
module is deactivated, but images can be taken from the realsense viewer.
The other laptop has usb 2.0 and I can't get a frame. I haven't tried all 3
mods at the same time. Thanks

şunu yazdı:

MartyG-RealSense, 2021-12-12:
I asked whether you had all three stream types enabled at the same time because when this is done, the enabling of the Motion Module can cause **Frames Not Received** to be displayed in the RealSense Viewer.  So if you are not enabling three streams including the Motion Module then that is fine.

 On a USB 2.0 connection a 400 Series camera has only a limited number of resolution and FPS settings available compared to when the camera is used on USB 3.  This is described in the documentation in the link below.

https://www.intelrealsense.com/usb2-support-for-intel-realsense-technology/

You should be able to achieve a depth stream of 640x480 resolution at 30 FPS on a USB 2 connection on D435i though, as demonstrated in the image below from my own D435i.

[image]

Can you confirm whether you are using the official 1 meter USB cable supplied with the camera or a longer USB cable of your own choice, please?

closed 2021-12-25 by MartyG-RealSense