realsenseai/librealsense #10049: GPIO stream not available ?

Misuliaut, 2021-12-12:
----------------------------------------------------------------------------------------------------

|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.49.0                      | 
| Language                            |  C#                       | 
| Segment			|  Robot                   | 

### Issue Description

I'm trying to read GPIO data from the camera (D435). To do this I call config.EnableStream(Stream.Gpio), but I get "Couldn't resolve requests" when starting the pipeline.

Is this stream supported at all ?

MartyG-RealSense, 2021-12-12:
Hi @Misuliaut  A RealSense team member advised in https://github.com/IntelRealSense/librealsense/issues/3001#issuecomment-453665014 in January 2019 that the GPIO pins were not programmable by the public.  

Recently, a RealSense user in https://github.com/IntelRealSense/librealsense/issues/9810 in October 2021 also tried to access the pins on a D435 using the C++ equivalent instruction **config.enable_stream(RS2_STREAM_GPIO1** and likewise received a 'Couldn't resolve requests' message.

MartyG-RealSense, 2021-12-19:
Hi @Misuliaut  Do you require further assistance with this case, please?  Thanks!

Misuliaut, 2021-12-19:
No thank you. We wanted to use the gpio for synchronization between the camera and another (not camera) device, but it seems we will have to find other ways to solve this.

closed 2021-12-19 by Misuliaut