realsenseai/librealsense #10050: bag

EMIIA, 2021-12-13:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:569

```

MartyG-RealSense, 2021-12-13:
Hi @EMIIA  Are you launching the RealSense Viewer program without a camera plugged in and attempting to load a bag file into the Viewer, please?

EMIIA, 2021-12-13:
https://user-images.githubusercontent.com/22911965/145867620-37726f2c-a76c-4762-acba-f9edef5a2469.png

MartyG-RealSense, 2021-12-14:
Closed as a duplicate of case https://github.com/IntelRealSense/librealsense/issues/10056

closed 2021-12-14 by MartyG-RealSense

EMIIA, 2021-12-14:
Still nothing works !!

MartyG-RealSense, 2021-12-14:
Please continue the discussion at https://github.com/IntelRealSense/librealsense/issues/10056 so that we are not discussing the same problem in two different locations.  Thanks!