realsenseai/librealsense #10051: L515 reduced FPS

wilsonz91, 2021-12-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.05.08.01| 
| Operating System & Version | Ubuntu 18  | 
| Kernel Version (Linux Only)    |  5.4                                         | 
| Platform                                 | PC   |
| SDK Version                            |  2.50.0                          | 
| Language                            |  NA (realsense viewer)                        | 
| Segment			|  Robot                          | 

### Issue Description
I'm using the L515 and notice that my rgb FPS is capped at ~16 Hz on the hardware (from realsense-viewer metadata). 
Things I've tried but the FPS remains stuck at ~16Hz:
1.  using the lowest resolution 
2. rgb streaming only (motion and depth disabled) 
3. I was originally using the official cable; tried using a D455 cable as well but same problem

MartyG-RealSense, 2021-12-13:
Hi @wilsonz91   If you have Auto-Exposure enabled, is RGB able to stream at 30 FPS if you disable an option in the RGB section of the Viewer controls called **Auto-Exposure Priority**

When Auto-Exposure is enabled, disabling Auto-Exposure Priority should enforce a constant FPS instead of permitting FPS to vary.  Auto-Exposure Priority can be disabled by left-clicking on the blue box beside the option (which indicates the On state) to change the box to black (Off).

For both the L515 and D455 camera models, the Auto-Exposure Priority option is under the **Controls** sub-category of the RGB section.

[image]

wilsonz91, 2021-12-13:
Hi @MartyG-RealSense , thanks, it works after making the changes. Would you know how do I do the same thing for the ROS launchfile?

MartyG-RealSense, 2021-12-13:
If you were using a D455 then the rosparam code below could be used in the launch file.

```
<rosparam>
/camera/rgb_camera/rs435_color_auto_exposure_priority: false
</rosparam>
```

In regard to L515, I am not aware of a previous reference demonstrating setting the auto-exposure priority status for that camera model.  However, a parameter name used for rosparam should be the same parameter name used for the rosrun rqt_reconfigure command and vice versa.

In regard to the path to the parameter, Intel documentation for the RealSense ROS wrapper in the link below states that L515 also uses **/camera/rgb_camera/** for the color stream.

https://dev.intelrealsense.com/docs/ros-wrapper#section-l-500-lidar-depth-cameras

wilsonz91, 2021-12-13:
Thanks @MartyG-RealSense, it works by setting it as follows: 
`<rosparam>
    /camera/rgb_camera/auto_exposure_priority: false
  </rosparam>`

MartyG-RealSense, 2021-12-13:
Great, thanks very much @wilsonz91 for the confirmation of the instruction for setting Auto-Exposure Priority that works on L515!

closed 2021-12-27 by MartyG-RealSense