realsenseai/librealsense #10053: Access Violation

EMIIA, 2021-12-13:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

EMIIA, 2021-12-13:
[image]

MartyG-RealSense, 2021-12-14:
Hi @EMIIA  There was a recent case of this error at https://github.com/IntelRealSense/librealsense/issues/9741 that was resolved by the RealSense user downloading and running an older version of the RealSense Viewer program.  The current Viewer version then started working normally.

Please go to the 'Assets' file list of the older **2.48.0** version at the link below and download and run the file called **Intel.RealSense.Viewer.exe**.  This is a standalone version of the Viewer that can be launched with a double-leftclick from the computer folder location that it is downloaded to (the launch shortcut icon for the Viewer on the Windows desktop will not launch this file).

https://github.com/IntelRealSense/librealsense/releases/tag/v2.48.0

After the standalone 2.48.0 Viewer is run, does this correct the Access Violation error with the current Viewer version launched with the desktop shortcut, please?

closed 2021-12-27 by MartyG-RealSense