realsenseai/librealsense #10054: Run box_dimensionaler_multicam without a chessboard

HyeYumKim, 2021-12-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455 | 
| Platform                                 | PC  |
| Language                            |  python  | 

Hello
I am measuring the volume using box_dimensionaler_multicam as python. After storing the corner coordinate values of the chessboard, we will load the coordinate values from the same location and run them next time.

I would like to know why the value when coordinate points2D is output in the presence of a chessboard board and the points2D stored and output power of the chessboard board come out differently.

<Run with the chessboard on.>
[image]
<>
[image]

MartyG-RealSense, 2021-12-14:
Hi @HyeYumKim  In your previous question on this subject in https://github.com/IntelRealSense/librealsense/issues/9951#issuecomment-966039933 it was recommended that if you are using a single camera then the intrinsics and extrinsics of the camera could be stored to enable use of the program without a chessboard.  Have you tried this approach, please?

Comparing your two sets of Points2D results, they do not seem to be very different.  Does using the set of saved-and-loaded values make a noticeable difference to the measurements of the volume?

HyeYumKim, 2021-12-15:
As you said then, the value was hard-coded and executed. When hard-coded, it catches completely different places from the object.
Is there a reason?

MartyG-RealSense, 2021-12-15:
The chessboard is designed to enable multiple cameras to calibrate their positions in relation to each other when placed around the outside of the chessboard.  So when a single camera is used, you should be able to use the intrinsics and extrinsics of that one camera instead of having to have the chessboard present to calibrate the camera against.

In your opening message, you said that you were "storing the corner coordinate values of the chessboard".  Could you explain more about this please, as it does not sound as though you are storing the intrinsic and extrinsic values of the camera.  Thanks very much.

HyeYumKim, 2021-12-16:
[image]

I want to get only the corner coordinates of the chessboard board, not the internal and external values of the camera.

What I understand is that the camera corrects the corner coordinates taken. So I think we only need to know the coordinate values of the corners taken by each camera.

Did I get it wrong? Please tell me if there's anything wrong.
Thank you.

MartyG-RealSense, 2021-12-16:
The title of this issue 'Run box_dimensionaler_multicam without a chessboard' suggested to me that you want to measure the volume of an object without using a calibration chessboard.  I apologize if I have misunderstood what you are aiming to achieve.
 
I went back to the start of the case to read it again.  It sounds as though you want to store the corner coordinates of the chessboard at the first time that you take a measurement and then take the chessboard away so that you can perform subsequent measurement sessions with the stored corner values without having to have the chessboard present.

For this to work, the camera would likely have to be fixed in the same position all the time and the measured object would have  to be placed at roughly the same position in front of the camera each time, because the program would not be able to use the chessboard to calibrate the camera to the board position if the board was not there or the program had been altered so that it does not perform the initial chessboard calibration phase.

HyeYumKim, 2021-12-17:
Yes, that’s right.
I'll try a few more times.
Thank you so much for your help!

MartyG-RealSense, 2021-12-17:
You are very welcome, @HyeYumKim - I look forward to your next report.  Good luck!

MartyG-RealSense, 2022-01-11:
Case closed due to no further comments received.

closed 2022-01-11 by MartyG-RealSense

raivil, 2022-03-09:
@HyeYumKim did you have success with removing the chessboard calibration? Can you share what was done and how you achieved it?

dreamerinthe, 2022-06-03:
Hi everyone,
I want to ask a question. I want to measure objects without a chessboard with a camera. Where should I save the information you mentioned here or where should I make changes in the python code? Can you help me?

MartyG-RealSense, 2022-06-03:
Hi @dreamerinthe  I believe that @HyeYumKim would be the best person to provide advice on this subject, because as far as I know they are the only RealSense user who has attempted this boardless adaptation and made progress with programming it,

dreamerinthe, 2022-06-06:
Hi dear @HyeYumKim can you help us?