realsenseai/librealsense #10056: 888

EMIIA, 2021-12-14:
| | |
|---|---|
|**librealsense**|2.48.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:569

![image](https://user-images.githubusercontent.com/22911965/145970598-0d367919-3337-436a-ade1-4813b6451661.png)

```

MartyG-RealSense, 2021-12-14:
Hi @EMIIA   In https://github.com/IntelRealSense/librealsense/issues/10053#issuecomment-993256941 it was suggested to you to install the 2.48.0 version of the standalone Viewer to try to correct an Access Violation error in the 2.49.0 Viewer.  As the version here is 2.48.0, it looks as though you followed that advice.  Is that correct, please?

After having installed the 2.48.0 Viewer, do you still experience the Access Violation when launching the **2.49.0** Viewer using the launch shortcut icon on the Windows desktop?

MartyG-RealSense, 2021-12-14:
You said in https://github.com/IntelRealSense/librealsense/issues/10050 that the camera still was not working.

closed 2021-12-27 by MartyG-RealSense