realsenseai/librealsense #10057: Python 3.10 wrapper support.

arsanysamuel, 2021-12-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Arch Linux  | 
| Kernel Version (Linux Only)    |  5.10.84-1-lts  | 
| Platform                                 |  PC  |
| SDK Version                            |  2.50  | 
| Language                            |  Python 3.10  | 

### Issue Description
I updated python from 3.9 to 3.10 and pip to 21.3.1, I tried to reinstall pyrealsense2 with pip and this error shows:
```
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
```

Will there be a pip package for 3.10?

MartyG-RealSense, 2021-12-14:
Hi @arsanysamuel  At the time of writing this, Python 3.9 is the most recent version supported by pip install.  The pyrealsense2 wrapper can built from source code for Python versions unsupported by pip install though.  The wrapper can either be built separately or at the same time as the librealsense SDK is built from source with CMake.

I do not have information regarding future plans and implementation timescales for pip install support of Python 3.10.  If you need to use pip, there was a very detailed discussion of workarounds for unsupported Python versions in https://github.com/IntelRealSense/librealsense/issues/8229 where a RealSense user suggested updating the **whl** file for newer versions using a hex editor at https://github.com/IntelRealSense/librealsense/issues/8229#issuecomment-767115385

MartyG-RealSense, 2021-12-21:
Hi @arsanysamuel  Do you require further assistance with this case, please?  Thanks!

arsanysamuel, 2021-12-21:
@MartyG-RealSense Thanks for your assistance. 
I'm having some trouble trying to install the compiled wrapper on a virtual environment, the wrapper installs by default on the main environment.

MartyG-RealSense, 2021-12-21:
Hi @arsanysamuel  Are you using a Virtual Machine (VM)?  If you are, which VM software are you using (Oracle VirtualBox, VMWare WorkStation Player, etc).

arsanysamuel, 2021-12-26:
@MartyG-RealSense Thanks for replying, I'm using Arch Linux as my main operating system with Python 3.10.
I compiled librealsense from source but pyrealsense2 was installed on the main python environement, how can I install it on a virtual environment?

MartyG-RealSense, 2021-12-26:
My understanding is that in a situation like this, the Python targetting problem would be handled by defining a **symbolic link (symlink)** like https://github.com/IntelRealSense/librealsense/issues/1657#issuecomment-386550240 describes for pyrealsense2

closed 2022-01-07 by MartyG-RealSense