realsenseai/librealsense #10059: when I use realsense-viewer in Raspberry 4B, no device connected

beyond21299, 2021-12-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 }                                                   | 
| Firmware Version                   | (v2.31.0)                                                  | 
| Operating System & Version |  Linux raspberrypi 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                                |  python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<when I use realsense-viewer ,no device connected>
I successfully set up the environment according to the official website article.
The article is https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md
when I use 'lsusb' command ,I can see this result as follow
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 17ef:608c Lenovo 
Bus 001 Device 004: ID 17ef:608d Lenovo 
Bus 001 Device 003: ID 03e7:2150 Intel Myriad VPU [Movidius Neural Compute Stick]
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

when I use python3 and import pyrealsense2 ,It's worked.
But when I use realsense-viewer ,no device connected.

MiSta77, 2021-12-22:
I'm using two identical Raspberry PI4, on one there is the same problem "'please connect a realsense camera...'", on the second not. On both kernel: Linux MyTestPi 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux

I can't see any difference, but found different behavior on the working Raspberry. After boot, lsusb shows 
"Bus 001 Device 006: ID 03e7:2150 Intel Myriad VPU [Movidius Neural Compute Stick]"

Then, after successfull starting the realsense-viewer, lsusb shows
"Bus 002 Device 002: ID 8087:0b37 Intel Corp." intead  "Intel Myriad VPU..."

On the Raspberry with the problem "please connect a realsense camera..." lsusb still lists the device "Intel Myriad VPU ...." when realsense-viewer was started.

caoluming, 2022-01-21:
你好，我也按照这篇文章步骤来安装的，版本跟你的一样，但中间一直有一些指令不正常，请问你是一条一条执行的吗？

closed 2026-10-23 by MartyG-RealSense