realsenseai/librealsense #10060: Landmark detection using depth data

rajeshjoshi8792, 2021-12-15:
im working on face authentication. and i can able to authenticate face using RGB image. but now i want to authenticate face using depth data. so is there any way that i can obtain facial landmark directly from depth data. i dont want to use RGB data for landmark detection.,

MartyG-RealSense, 2021-12-15:
Hi @rajeshjoshi8792  I researched your question extensively, but most facial landmark detection systems that came up in my research involved use of RGB in generating landmarks, with depth used for anti-spoofing measures.  I did though locate a research paper that discusses making use of RealSense D435 and depth-only data for facial analysis.

https://www.sciencedirect.com/science/article/pii/S1077314221000916

rajeshjoshi8792, 2021-12-15:
@MartyG-RealSense  How i use depth data for anti-spoofing measurements.?

MartyG-RealSense, 2021-12-15:
The RealSense SDK provides a C++ example for doing so with its **Dlib** face authentication wrapper, though thee example makes use of both RGB and depth data.

**RealSense Dlib wrapper**
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/dlib

**Dlib 'rs-face-dlib' example program**
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/dlib/face

rajeshjoshi8792, 2021-12-15:
@MartyG-RealSense  thank you for reply.  how can i use this code in python? is there python wrapper code available for anti spoofing?

MartyG-RealSense, 2021-12-15:
The Dlib face detection library was made for C++ but supports Python.  If you go to the link below and find the **Help / Info** section in the side-panel, expanding open the **Examples: Python** section shows a list of Dlib face sample programs for Python.

http://dlib.net/

[image]

A RealSense team member in another case also recommended the Dlib approach, suggesting in https://github.com/IntelRealSense/librealsense/issues/4430#issuecomment-511660804 to google for the term **facial-landmarks-dlib-opencv-python**

rajeshjoshi8792, 2021-12-15:
@MartyG-RealSense  i checked above links. but its using RGB image only. i want to use depth data.

MartyG-RealSense, 2021-12-15:
Another suggestion about anti-spoofing with depth information is given at https://github.com/IntelRealSense/librealsense/issues/3451#issuecomment-471420572

closed 2021-12-27 by MartyG-RealSense