realsenseai/librealsense #10061: get video frame from black&white D430 camera

agreyfox, 2021-12-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux ( debina 11 | 
| Kernel Version (Linux Only)    |  5.10.70-1                                     | 
| Platform                                 | PC  |
| SDK Version                            | 2}                          | 
| Language                            |  python }                          | 
| Segment			| others                           | 

### Issue Description
Quick question: Since my D430 camera is not RGB camera, How can I get video frame then I can use cv2.imshow to display the video content.  

if try to use  config.enable_stream(rs.stream.color) function , and pipleline.start(config), It report error like 
```
>>> pipeline.start(config)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Couldn't resolve requests
```
any suggestion with  D430 camera?

Thanks , Replay and help will be very appreciate!

agreyfox

MartyG-RealSense, 2021-12-15:
Hi @agreyfox  The D415 and D455 camera models can stream RGB from the left infrared sensor, but the D430 and D435 / D435i only have Y8 / Y16 infrared available from the left infrared sensor.

You could obtain a monochrome RGB-like image by streaming Y8 Infrared from the left infrared sensor.  An example of a pyrealsense2 script shared by a RealSense user for displaying the Infrared stream with cv2.imshow is below.

```
import pyrealsense2 as rs
import numpy as np
import cv2
# We want the points object to be persistent so we can display the 
#last cloud when a frame drops
points = rs.points()
# Create a pipeline
pipeline = rs.pipeline()
#Create a config and configure the pipeline to stream
config = rs.config()
config.enable_stream(rs.stream.infrared, 1, 1280, 720, rs.format.y8, 30)
# Start streaming
profile = pipeline.start(config)
# Streaming loop
try:
while True:
# Get frameset of color and depth
frames = pipeline.wait_for_frames()
ir1_frame = frames.get_infrared_frame(1) # Left IR Camera, it allows 1, 2 or no input
image = np.asanyarray(ir1_frame.get_data())
cv2.namedWindow('IR Example', cv2.WINDOW_AUTOSIZE)
cv2.imshow('IR Example', image)
key = cv2.waitKey(1)
# Press esc or 'q' to close the image window
if key & 0xFF == ord('q') or key == 27:
cv2.destroyAllWindows()
break
finally:
pipeline.stop()
```

agreyfox, 2021-12-17:
Hi MartyG, Thanks for your information, it is very helpful!.

closed 2021-12-17 by agreyfox

MartyG-RealSense, 2021-12-17:
I'm pleased that the information was helpful to you, @agreyfox - thanks very much for the update!