realsenseai/librealsense #10062: No Frames Recieved #L515 #WINDOW10

ksirirux, 2021-12-15:
**RealSense L515 with window10** show  No Frames Received!

[image]

I buy realsense from ebay but I can't contact seller. 
Please advise me?

ksirirux, 2021-12-15:
System Information:

[image]

ksirirux, 2021-12-15:
my system information:

[image]

ksirirux, 2021-12-15:
Error when first connect

[image]

MartyG-RealSense, 2021-12-15:
Hi @ksirirux  Please first try unplugging the micro-sized end of the USB cable from the base of the camera, turning the connector around the opposite way and re-inserting it into the camera (USB-C cables are two-way insertion at the micro end).  Does reversing the connector resolve your problem?

ksirirux, 2021-12-15:
I reverse USB-C camera is not detect ...
[image]

MartyG-RealSense, 2021-12-15:
Thank you for testing the orientation of the USB connector.  Next, please try updating the firmware driver of your camera from version 1.5.5.0 to **1.5.8.1**.

To do this, first turn the cable orientation back to its previous position so that the camera can be detected in the RealSense Viewer.  Then visit the L515 firmware releases page at the link below and download the firmware file **Signed_Image_UVC_1_5_8_1.bin** to your computer.

https://dev.intelrealsense.com/docs/firmware-releases-l500#section-l500-series-firmware-downloads

[image]

Once the firmware file is downloaded, launch the RealSense Viewer and go to the **More** option near the top of the Viewer's options side-panel.  Select the **Update Firmware** menu option to open a file navigation box and navigate to the folder location on your computer that the firmware file was downloaded to.  Select the file and the L515 should begin updating its firmware to version 1.5.8.1.

[image]

ksirirux, 2021-12-15:
Error show when connect device....

[image]

ksirirux, 2021-12-15:
Problem still.. in v1.05.08.01 

[image]

MartyG-RealSense, 2021-12-15:
Thanks very much for the information and images.  Next, could you test whether you can successfully run the **rs-capture** example program with your L515 in Windows using the instructions in https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

ksirirux, 2021-12-16:
@MartyG-RealSense  thank you for your support ....  

After run  rs-capture 
Error shown ... 

[image]

MartyG-RealSense, 2021-12-16:
Thank you.  Next, please try uninstalling and reinstalling the L515 camera drivers in Windows using the procedure below.

1.  Open the Windows **Device Manager**.  You can find the Device Manager by typing **device** into the text box at the bottom of your Windows screen, next to the corner button where you shut down the PC.

2.  Expand open the **Cameras** category of the Device Manager whilst the L515 is plugged in.  There should be two L515 drivers listed for RGB and Depth.

[image]

3.  Right-click on each of the two drivers and select the **Uninstall device** menu option.

[image]

4.  When both of the L515 drivers have been removed, unplug the camera, wait a couple of seconds and plug it back into the USB port.  The two drivers that were removed should now automatically re-install.

Is the depth stream now able to work normally after performing this procedure?

ksirirux, 2021-12-17:
Thank you so much for your advice.
I think the problem may cause from my usb 3.0 port.
I'll try to other new computer for test it again. 

[image]

MartyG-RealSense, 2021-12-17:
You are very welcome, @ksirirux - I look forward to your next report.  Good luck!

closed 2022-01-09 by MartyG-RealSense