realsenseai/librealsense #10064: LabVIEW 2020 SW startup problem on D455 model

woodal, 2021-12-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC  |
| SDK Version                            |         2.49.0.3474                 | 
| Language                            | labview 2020                           | 
| Segment			|  VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello there!

[image]

I
Opening the LabVIEW Example was successful. 
Opening LabVIEW Example by changing the dll file was successful.
However, the LabVIEW program runs but no screen appears. I have both LabVIEW 64bit and 32bit. Even if I try to run both by changing the dll file, the screen does not appear.
[image]

RealSense_HelloWorld_Left_Right_IR.vi is excutable.
[image]

Any help on how to solve this problem would be appreciated.
Thanks!!

MartyG-RealSense, 2021-12-15:
Hi @woodal  Although the **Monochrome + Depth** LabVIEW example is included in the LabVIEW package, it is not described in the wrapper documentation.  From the descriptive text on the window of the image of the example that you kindly provided though, it looks as though it is streaming monochrome color from the left infrared sensor, probably with the monochrome UYVY format.

[image]

D455 supports streaming color from the left infrared imager if firmware version **5.12.8.200** or newer is installed, so your 5.13.0.50 firmware should allow this.  I do not know if this specific LabVIEW example has been tested with D455 though, as it would have more likely been designed for the older D415 model (which also supports color from the left infrared sensor - the D435 / D435i models do not).

Does the **RealSense Hello World: Color + Depth** example function correctly with your D455?  That example also streams color from the left infrared sensor.  https://github.com/IntelRealSense/librealsense/issues/8886 also demonstrates a D455 being successfully used with the **Hello World: Point Cloud Plus** that makes use of color from left imager for its color image.

woodal, 2021-12-16:
Thank you @MartyG-RealSense !!

All programs recognize the D455 and run without error. 
However, not all programs except "HelloWorld_Left_Right_IR.vi" will play the screen.😭

> Hi @woodal Although the **Monochrome + Depth** LabVIEW example is included in the LabVIEW package, it is not described in the wrapper documentation. From the descriptive text on the window of the image of the example that you kindly provided though, it looks as though it is streaming monochrome color from the left infrared sensor, probably with the monochrome UYVY format.
> 
> [image]
> 
> D455 supports streaming color from the left infrared imager if firmware version **5.12.8.200** or newer is installed, so your 5.13.0.50 firmware should allow this. I do not know if this specific LabVIEW example has been tested with D455 though, as it would have more likely been designed for the older D415 model (which also supports color from the left infrared sensor - the D435 / D435i models do not).
> 
> Does the **RealSense Hello World: Color + Depth** example function correctly with your D455? That example also streams color from the left infrared sensor. #8886 also demonstrates a D455 being successfully used with the **Hello World: Point Cloud Plus** that makes use of color from left imager for its color image.
> 
> I am not aware of a previous case of the LabVIEW wrapper being used with LabVIEW 2020, though your successful launch of the **RealSense Hello World: Left-Right** example suggests that it is compatible.

woodal, 2021-12-16:
When I try again, some programs work.
The rest will be dll's problem, right?
[image]

MartyG-RealSense, 2021-12-16:
At the time of writing this, the latest release of the LabVIEW package was for the old SDK version 2.33.0.  We know that the wrapper works with newer SDK versions than 2.33.0, but there is no way to know for certain which samples will work with modern SDK dll library files until they are tried.

johnras, 2021-12-16:
Hi Woodal and MartyG

It works on my machine (64-bit) with RealSense API version 2.40.00 and firmware 05.12.09.00 (as shown in the pic below).  I don't believe that I made any changes to the LabVIEW code in this particular example (other than updating the API as described in my original post (#7920)).   
[image]

HOWEVER - when I update my firmware to 05.13.00.50 I have the same issue described by Woodal (see pic before) with errors in the Capture1 and Capture2 blocks.
[image]

If I revert backwards and start going from known working firmware (05.12.09.00) to successively newer firmware I get the following results (keeping API 2.40.0):
Firmware -       Functions?
05.12.10.00 -    Yes
05.12.11.00 -    Yes
05.12.12.100 -  Yes
05.12.13.50 -    Yes
05.12.14.50 -    Yes
05.12.15.50 -    No

Updated API to 2.49 in Labview folder (at least I think, I replaced the realsense2.dll and realsense2.lib files in the labview folder, but it still shows 24000 on the VI panel).
Firmware        Functions?
05.12.14.50     Yes
05.12.15.50     No

So, basically there appears to be something within firmware version 05.12.15.50 that is incompatible with this LabView example. Digging a little deeper, it appears the software times out before the frame arrives as there are no earlier reported errors.  If you have it working in one of the demos, it might be worth looking through the configuration portions of code to see what is different between the example code and the working one.

Or the easiest thing to do might be to just revert the firmware back a version or two...

MartyG-RealSense, 2021-12-17:
Thanks so much @johnras for your detailed advice to @woodal   :)

woodal, 2021-12-17:
Thanks so much @johnras and @MartyG-RealSense !
After changing Firmware, everything works fine.
thank you!😂😂

MartyG-RealSense, 2021-12-17:
That's great to hear, @woodal ... thanks again @johnras for your helpful advice!

closed 2021-12-24 by MartyG-RealSense