realsenseai/librealsense #10065: Disabling IR emitter of D435 in Python

ToshiEAB, 2021-12-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-80-generic                | 
| Platform                                 | PC  |
| SDK Version                            |  pyrealsense2              2.50.0.3812          | 
| Language                            |  python                        | 
| Segment			|  others                           | 

### Issue Description
I use RealSence D435 in both C++ and Python. I want to disable IR emitter. I don't have any issue with it in C++ but cannot do the same in Python. My Python codes are as follows:

import pyrealsense2 as rs
...
config = rs.config()
config.enable_stream(rs.stream.depth, 360, 640, rs.format.z16, 60)
config.enable_stream(rs.stream.color, 360, 640, rs.format.bgr8, 60)
self.pipeline = rs.pipeline()
profile = self.pipeline.start(config)
depth_sensor = profile.get_device().first_depth_sensor()
print(depth_sensor.supports(rs.option.emitter_enabled))   --> This gives "False"
depth_sensor.set_option(rs.option.emitter_enabled, 0) --> This gives "RuntimeError: object doesn't support option #18"

Does this mean I can disable the emitter in C++ but not in Python?

Thanks in advance.

MartyG-RealSense, 2021-12-15:
Hi @ToshiEAB  Does the alternative Python script below disable the emitter or does it still generate the **RuntimeError: object doesn't support option #18** error?

```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
pipeline_profile = pipeline.start(config)
device = pipeline_profile.get_device()
depth_sensor = device.query_sensors()[0]
if depth_sensor.supports(rs.option.emitter_enabled):
depth_sensor.set_option(rs.option.emitter_enabled, 0)
```

ToshiEAB, 2021-12-15:
Hi @MartyG-RealSense 

Thanks for answering my question. I tried your codes and still see the #18 error.

MartyG-RealSense, 2021-12-15:
Another way to do it is like this:

```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
pipeline_profile = pipeline.start(config)
device = pipeline_profile.get_device()
depth_sensor = device.query_sensors()[0]
emitter = depth_sensor.get_option(rs.option.emitter_enabled)
print("emitter = ", emitter)
set_emitter = 0
depth_sensor.set_option(rs.option.emitter_enabled, set_emitter)
emitter1 = depth_sensor.get_option(rs.option.emitter_enabled)
print("new emitter = ", emitter1)
```

***********************

If this does not work then an alternative method of disabling the IR emitter is to set the **Laser Power** option to a value of '0' using the **rs.option.laser_power** instruction.  A RealSense team member provides an example of Python scripting to do this at https://github.com/IntelRealSense/librealsense/issues/1258#issuecomment-374112555

ToshiEAB, 2021-12-16:
@MartyG-RealSense 
That code did not work either.  However, I found a solution. I had to update the firmware version to "05.11.15.00" recommended by realsense-viewer.

Thanks for your willingness to help me anyway.

closed 2021-12-16 by ToshiEAB

MartyG-RealSense, 2021-12-16:
It's excellent to hear that you found a solution, @ToshiEAB - thanks very much for the update!