realsenseai/librealsense #10066: cannot receive an infrared image #L515

jpoehlsACO, 2021-12-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   |01.05.08.01 | 
| Operating System & Version | WIN 10  | 
| Platform                                 | PC |
| SDK Version                            | v2.49.0                       | 
| Language                            |  python                        | 
| Segment			|  others       | 

### Issue Description
Hello, I would like to mapp the infrared image with the depth information. There are no problems reading out the depth information (and also colour information). However, I cannot read out the infrared image information. Does anyone have an idea?
 I have tried the different resolution levels.
When I call the code, I get the error:
`
AttributeError: 'bool' object has no attribute 'infrared_image'
`

`
import pyrealsense2 as rs
import numpy as np

class DepthCamera:
    def __init__(self):

        # Configure depth and color streams
        self.pipeline = rs.pipeline()
        config = rs.config()

       # config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
       # config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
        config.enable_stream(rs.stream.infrared,1024,768,rs.format.y8,30)

        # Start streaming
        self.pipeline.start(config)

    def get_frame(self):
        frames = self.pipeline.wait_for_frames()
        #depth_frame = frames.get_depth_frame()
        #color_frame = frames.get_color_frame()
        infrared_frame = frames.get_infrared_frame()

        #depth_image = np.asanyarray(depth_frame.get_data())
        #color_image = np.asanyarray(color_frame.get_data())
        infrared_image = np.asanyarray(infrared_frame.get_data())
        if not infrared_frame  :
            return False, None, None
        return True. infrared_image

    def release(self):
        self.pipeline.stop()`

MartyG-RealSense, 2021-12-15:
Hi @jpoehlsACO  As the error message relates to a bool and your code looks okay (though I would recommend setting the Infrared stream to the same 640x480 resolution as Depth), the line `return True. infrared_image` looks like a possible cause of the rror.  Could you try commenting that line out with // to see whether it resolves the error, please?

jpoehlsACO, 2021-12-15:
Hi @MartyG-RealSense , Thank you for the quick reply. The error was indeed hidden in the line. The full stop had to be replaced by a comma. I should have recognised that myself. Many thanks for the good support.

Here the working code:

`import pyrealsense2 as rs
import numpy as np

class DepthCamera:
    def __init__(self):

        # Configure depth and color streams
        self.pipeline = rs.pipeline()
        config = rs.config()

        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
        config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
        config.enable_stream(rs.stream.infrared,640,480,rs.format.y8,30)

        # Start streaming
        self.pipeline.start(config)

    def get_frame(self):
        frames = self.pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        infrared_frame = frames.get_infrared_frame()

        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        infrared_image = np.asanyarray(infrared_frame.get_data())
        #if not infrared_frame  :
        #    return False, None, None
        #return True. infrared_image
        return depth_frame, color_image,infrared_image

    def release(self):
        self.pipeline.stop()`

closed 2021-12-15 by jpoehlsACO

MartyG-RealSense, 2021-12-15:
You're very welcome, @jpoehlsACO - it's great to hear that a solution was achieved.  Thanks very much for sharing your full L515 script with the RealSense community  :)