realsenseai/librealsense #10067: External RGB and Realsense 455 align

codesian, 2021-12-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455 | 
| Firmware Version                   | 2.45.0 | 
| Operating System & Version |   Linux Ubuntu 18  | 
| Kernel Version (Linux Only)    |    4.9.201-tegra                                         | 
| Platform                                 |  NVIDIA Jetson  |
| SDK Version                            |  2.45.0                          | 
| Language                            |  C++                         | 
| Segment			|  AR                            | 

Hi! Im trying to use an external RGB sensor and align with realsense depth. Realsense 455 RGB sensor seems a little bit small in terms of resolution and detail for us, we need something bigger. We are using a test Sony IMX sensor, readed with VideoCapture in OpenCV. Of course before do some code, we calculate intrinsics and extrinsics with realsense camera in a rig. Then we used Realsense rs-software-device to get frames from both realsense and sony sensor, so far so good. But we stuck at align. RGB image seems correct, but aligned depth seems full black. Sorry code its untidy.

```

#include <iostream>
#include <thread>
#include <chrono>
#include <librealsense2/rs.hpp>
#include <librealsense2/hpp/rs_internal.hpp>
#include <opencv4/opencv2/opencv.hpp>
#include <opencv4/opencv2/core/mat.hpp>
#include <iomanip>

cv::VideoCapture capture;

struct synthetic_frame
{
    int x, y, bpp;
    std::vector<uint8_t> frame;
};

class custom_rgb_source
{
private:

    synthetic_frame color_frame;
    std::chrono::high_resolution_clock::time_point last;
    
public:

    custom_rgb_source() //: capture(0)
    {

        capture.set(cv::CAP_PROP_FRAME_WIDTH, 2048);
        capture.set(cv::CAP_PROP_FRAME_HEIGHT, 1536); 
        color_frame.x = 2048;
        color_frame.y = 1536;
        color_frame.bpp = 3;
        last = std::chrono::high_resolution_clock::now();
    }

    synthetic_frame& get_rgb_image(){
        
        auto now = std::chrono::high_resolution_clock::now();
        cv::Mat frame;        
        capture.read(frame);
        std::vector<uint8_t> pixels_color;
        
        if (frame.isContinuous()) {
            pixels_color.assign(frame.data, frame.data + frame.total() * frame.channels());
        }else{
            for (int i = 0; i < frame.rows; ++i) {
                pixels_color.insert(color_frame.frame.end(), frame.ptr<uint8_t>(i), frame.ptr<uint8_t>(i)+frame.cols*frame.channels());
            }
        }

        color_frame.frame = std::move(pixels_color);
        return color_frame;

    }

    rs2_intrinsics create_intrinsics()
    {
        rs2_intrinsics intrinsics = { 
            
           
            2048, 1536,
            864.3000, 824.7314,
            1514.8800346232729,1514.8800346232729,
            RS2_DISTORTION_BROWN_CONRADY ,{ 0.0029763750419588564,-0.00069578419440207275,0.00040522151733745005,0.00049985637962198025,-0.0084025276764905944 } };

        return intrinsics;
    }

};

int main(int, char**) {

    if(!capture.open(3)){}

    int frame_number = 0;
    custom_rgb_source rgb_source;
    rs2::context ctx;
    rs2::device d;
    rs2::syncer syncer(3);
    std::vector<rs2::sensor> sensors;
    rs2::stream_profile alignprofile;
    rs2::frame_queue q_color(1);
    rs2::frame_queue q_depth(1);

    bool profileFinded = false;

    for (auto dev : ctx.query_devices()) {            
        
        for (rs2::sensor sensor : dev.query_sensors()) {
                    
                    profileFinded = false;
                    rs2::stream_profile dprofile;

                    auto sensor_stream_profiles = sensor.get_stream_profiles();
                    if (sensor.is<rs2::depth_sensor>()) {
                        dprofile = sensor_stream_profiles[215];
                        alignprofile = dprofile;
                        sensor.open(dprofile);
                        sensors.push_back(sensor);
                    }
        }
        break;   
    }

    rs2_intrinsics color_intrinsics = rgb_source.create_intrinsics();
    rs2::software_device ndev;
    auto color_sensor = ndev.add_sensor("EXT"); 
    auto color_stream = color_sensor.add_video_stream({  RS2_STREAM_COLOR, 0, 0, 2048,
                                1536, 10, 3,
                                RS2_FORMAT_RGB8, color_intrinsics });

    ndev.create_matcher( RS2_MATCHER_DIC );  
    color_sensor.open(color_stream);
    sensors.push_back(color_sensor);
    alignprofile.register_extrinsics_to(color_stream, { { 0.9999,0.0106,0.0034,-0.0105,0.9997,-0.0240,-0.0033,0.0240,0.9997 },{ -139.9843,-0.4026,-7.4355 } });

    rs2::frame_queue q(2);
    std::vector<rs2::frame> bundle;

    rs2::processing_block bundler([&](rs2::frame f, rs2::frame_source& src) {
        bundle.push_back(f);     //add frame to vector
        if (bundle.size() == 2){ //or however many when full
            auto fs = src.allocate_composite_frame(bundle);
            src.frame_ready(fs);
            bundle.clear();
        }
    });

    bundler.start(q); 

    for(rs2::sensor s : sensors){
        s.start([&](rs2::frame f){
            bundler.invoke(f);
        });
    }

    while (true) {
            
            synthetic_frame& ncframe = rgb_source.get_rgb_image();
            double ts = (std::chrono::duration_cast<std::chrono::nanoseconds>(std::chrono::system_clock::now().time_since_epoch()).count() / 1000000.0);
            rs2_time_t timestamp = (rs2_time_t) ts ;
            auto domain = RS2_TIMESTAMP_DOMAIN_GLOBAL_TIME;

            color_sensor.on_video_frame( { ncframe.frame.data(),
                                       []( void * ) {},
                                       ncframe.x * ncframe.bpp,
                                       ncframe.bpp,
                                       timestamp, domain, frame_number,
                                       color_stream });
           
            ++frame_number;

            rs2::frameset fs = q.wait_for_frame();

            int ready = 0;
            for (const rs2::frame& f : fs){
                if( f.get_profile().stream_type()  == RS2_STREAM_DEPTH){
                    ready += 1; 
                }else if( f.get_profile().stream_type()  == RS2_STREAM_COLOR){
                    ready += 3;
                }
            }

            if(fs.size() == 2  && ready == 4 ){  
                    
                rs2::video_frame depthFrame = fs.first_or_default(RS2_STREAM_DEPTH);
                rs2::video_frame rgbFrame   = fs.first_or_default(RS2_STREAM_COLOR);
                
                cv::Mat df = cv::Mat(depthFrame.get_height(), depthFrame.get_width(), CV_16UC1, (void *) depthFrame.get_data(), cv::Mat::AUTO_STEP);
                cv::Mat rf = cv::Mat(rgbFrame.get_height(), rgbFrame.get_width(), CV_8UC3, (void *) rgbFrame.get_data(), cv::Mat::AUTO_STEP);
                
                for (const rs2::frame& f : fs){
                    std::cout << std::setprecision (20) << f.get_profile().stream_name() << " TS:" << f.get_timestamp() << " DOM: " << f.get_frame_timestamp_domain() <<  " FN: " << f.get_frame_number() << " | ";
                }
                
                rs2::align align(RS2_STREAM_COLOR);
                rs2::frameset aligned = align.process(fs);  
            
                if(aligned.size() == 2){   
 
                    rs2::video_frame afd = aligned.first_or_default(RS2_STREAM_DEPTH);
                    rs2::video_frame afc = aligned.first_or_default(RS2_STREAM_COLOR);
                    cv::Mat d = cv::Mat(afd.get_height(), afd.get_width(), CV_16UC1, (void *) afd.get_data(), cv::Mat::AUTO_STEP).clone();
                    cv::Mat r = cv::Mat(afc.get_height(), afc.get_width(), CV_8UC3, (void *) afc.get_data(), cv::Mat::AUTO_STEP).clone();
                     
                }

                std::this_thread::sleep_for(std::chrono::milliseconds(200));
            
            }else{
                std::cout << "NO " << ready << std::endl;
            }
 

	}

    return 0;
    

}

```

Thanks for your help

MartyG-RealSense, 2021-12-16:
Hi @codesian  In https://github.com/IntelRealSense/librealsense/issues/4202#issuecomment-502722403 a RealSense team member provides advice for information to include in the script when aligning RealSense depth with the 4K RGB of an external camera.

codesian, 2021-12-16:
Sure, as i mention on my issue, i calculate RGB sensor instrinsics

```
 2048, 1536,
            864.3000, 824.7314,
            1514.8800346232729,1514.8800346232729,
            RS2_DISTORTION_BROWN_CONRADY ,{ 0.0029763750419588564,-0.00069578419440207275,0.00040522151733745005,0.00049985637962198025,-0.0084025276764905944 } };

```

and extrinsics with realsense depth camera (IR imager), 

```
alignprofile.register_extrinsics_to(color_stream, { { 0.9999,0.0106,0.0034,-0.0105,0.9997,-0.0240,-0.0033,0.0240,0.9997 },{ -139.9843,-0.4026,-7.4355 } });

```

using MatLab calibration to obtain values. Before wrote this issue i read all issues available at realsense.

MartyG-RealSense, 2021-12-16:
My knowledge of software-device programming is admittedly limited, so thanks in advance for your patience.  You may have also seen the case https://github.com/IntelRealSense/librealsense/issues/1103 about depth-color alignment with an external camera during your research of cases involving software-device.

My understanding is that **ctx.query_devices** should add **any** detected camera to its list, including a non-RealSense "platform camera" such as your Sony IMX.  And you have mentioned that your script is able to access color data from the Sony camera, and just has problems with the alignment aspect of the project.

You are also setting the window size to 2048x1536, which is the color resolution that your Sony camera is using for RGB?

MartyG-RealSense, 2021-12-16:
I would suspect that it is related to an issue somewhere in the align block of code shown below, if you are still using that code.

[image]

A discussion in the link below features someone who was trying to perform depth-color alignment with OpenCV code in librealsense and could achieve depth and color individually but only had a grey screen after performing alignment.

https://stackoverflow.com/questions/51185706/mapping-depth-image-to-color-image-with-realsense2-about-opencv

codesian, 2021-12-16:
Sorry but i think its not related, 

In this case, user got a grey image just because it`s reading a depth image as RGB, 

```
 Mat other_frameaM(Size(width, height), CV_8UC3, (void*)other_frame.get_data(), Mat::AUTO_STEP);
 Mat aligned_depthM(Size(width, height), **CV_8UC3**, (void*)aligned_depth_frame.get_data(), Mat::AUTO_STEP);
```
in my case 

```
 cv::Mat df = cv::Mat(depthFrame.get_height(), depthFrame.get_width(), **CV_16UC1**, (void *) depthFrame.get_data(), cv::Mat::AUTO_STEP);
                cv::Mat rf = cv::Mat(rgbFrame.get_height(), rgbFrame.get_width(), CV_8UC3, (void *) rgbFrame.get_data(), cv::Mat::AUTO_STEP);

```

Depth image must be read as CV_16UC1 not CV_8UC3.  In my case image its full black. Ok, lets think for a moment that its an opencv problem, i remove it from the equation, printing rs2::frame content on std::cout an all values are 0. I Think that problem its on align function.

Thanks for your time

MartyG-RealSense, 2021-12-17:
I researched your problem extensively but there is little information available in regard to having zero depth values after using alignment with software-device.  There was a case in the link below that seemed to have experienced a similar issue though.

https://support.intelrealsense.com/hc/en-us/community/posts/360037518133-Software-device-depth-image-aligning-to-rgb

codesian, 2021-12-22:
Sorry but i think its not related.
I can´t go so far, im checking calibration between sensors but all seems correct. I think that problem its on align function.

MartyG-RealSense, 2021-12-22:
Unfortunately there are very few available references about software-device **and** alignment and I have covered those few references, so there is nothing else that I can suggest.

MartyG-RealSense, 2021-12-28:
Hi @codesian  Do you have an update about this case that you can provide, please?  Thanks!

codesian, 2021-12-28:
No, we are still working and testing. We are a little lost with this.

MartyG-RealSense, 2021-12-29:
Okay, please do let me know if you make progress from your tests.  Good luck!

MartyG-RealSense, 2022-01-10:
Hi @codesian  Do you have an update about the tests that you have been performing that you can provide, please?  Thanks!

codesian, 2022-01-10:
No way, seems a problem aligning external RGB image and depth.

MartyG-RealSense, 2022-01-10:
I researched the problem further and located another software-device case at https://github.com/IntelRealSense/librealsense/issues/6522 where the depth and RGB images were fine individually but alignment did not work.

A RealSense team member offered advice in https://github.com/IntelRealSense/librealsense/issues/6522#issuecomment-639044215

And https://github.com/IntelRealSense/librealsense/issues/6522#issuecomment-641027578 contains the eventual script solution of the

closed 2022-01-22 by MartyG-RealSense