realsenseai/librealsense #10068: Global Time for motion_sensor / imu is in the future relative to system clock

jwpleow, 2021-12-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i, D455 | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-91                                        | 
| Platform                                 | PC / x86-64 / Intel CPU |
| SDK Version                            | v2.50.0                        | 
| Language                            |  C++                     | 
| Segment			|  Robot                      | 

### Issue Description
Hello!
I notice when querying the motion sensor's global timestamps from both D435i and D455s, I get timestamps that vary from a 4ms delay to 40ms in the future. Is this expected behaviour?

Sample code (adapted from examples/motion)
```
#include <librealsense2/rs.hpp>
#include <cstring>
#include <thread>
#include <chrono>
#include <iostream>

int main(int argc, char * argv[]) try
{
    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;

    // Add streams of gyro and accelerometer to configuration
    cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F);
    cfg.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F);

    // Start streaming with the given configuration;
    // Note that since we only allow IMU streams, only single frames are produced
    auto profile = pipe.start(cfg, [&](rs2::frame frame)
    {
        // Cast the frame that arrived to motion frame
        auto motion = frame.as<rs2::motion_frame>();
        // If casting succeeded and the arrived frame is from gyro stream
        if (motion && motion.get_profile().stream_type() == RS2_STREAM_GYRO && motion.get_profile().format() == RS2_FORMAT_MOTION_XYZ32F)
        {
            // Get the timestamp of the current frame
            double ts = motion.get_timestamp();
            auto tsUs = static_cast<int64_t>(ts) * 1000;
            int64_t currSystemStamp = std::chrono::duration_cast<std::chrono::microseconds>(std::chrono::system_clock::now().time_since_epoch()).count();

            std::cout << "System clock: " << currSystemStamp << std::endl;
            std::cout << "Gyro clock: " << tsUs << std::endl;
            std::cout << "Difference: " << currSystemStamp - tsUs << std::endl;

            // // Get gyro measures
            // rs2_vector gyro_data = motion.get_motion_data();
        }
        // If casting succeeded and the arrived frame is from accelerometer stream
        if (motion && motion.get_profile().stream_type() == RS2_STREAM_ACCEL && motion.get_profile().format() == RS2_FORMAT_MOTION_XYZ32F)
        {
            // // Get accelerometer measures
            // rs2_vector accel_data = motion.get_motion_data();
        }
    });

    // Main loop
    while (true)
    {
        std::this_thread::sleep_for(std::chrono::seconds(1));
    }
    // Stop the pipeline
    pipe.stop();

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

MartyG-RealSense, 2021-12-16:
Hi @jwpleow  A noticable drift in timing can sometimes occur when Global Time is enabled.  Usually, the only solution is to disable Global Time.  Information about doing so in C++ is provided at https://github.com/IntelRealSense/librealsense/issues/9076

If having Global Time enabled is important to you then you could check whether adding a **hardware_reset()** instruction to reset the camera when the script launches improves your results.  A C++ code example for this is at https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

jwpleow, 2021-12-17:
I see, thank you!

closed 2021-12-17 by jwpleow

chenxiaocongAI, 2023-08-28:
@jwpleow  do you success solve this problem