realsenseai/librealsense #10070: jetson nano pyrealsense2 RuntimeError: Frame didn't arrive within 5000

MAKER-park, 2021-12-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  Linux jetbot-desktop 4.9.140-tegra #1 SMP PREEMPT Mon Dec 9 22:47:42 PST 2019 aarch64 aarch64 aarch64 GNU/Linux| 
| Platform                                 | nvidia jetson nano  |
| SDK Version                            |  { 2.50.0 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

hello! i'm using my jetson nano (with jetbot)  and intel realsense d415 camera

[image]
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py
first example is working fine!
looks like it working good!

but next example using with opencv.... it have some problem... 

jetbot@jetbot-desktop:~/Document/git_project/camera_test$ python3 test.py
Traceback (most recent call last):
  File "test.py", line 45, in <module>
    frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

i'm using this code
how can i do for slove this error?

thank you for read my problem!

MartyG-RealSense, 2021-12-16:
Hi @MAKER-park  It can be assumed from your success with **python-tutorial-1-depth.py** that the pyrealsense2 wrapper is working and the camera is able to stream depth data successfully.

python-tutorial-1-depth.py generates its image from text characters and does not use graphics.  So you could first try updating the graphics drivers of your Ubuntu computer to see whether it makes a positive difference.

MAKER-park, 2021-12-16:
Hi!! @MartyG-RealSense  Thank you so much for your reply.
But I know that the board I am using is not able to upgrade the gpu to jetson nano.

Is there any other way?

MartyG-RealSense, 2021-12-16:
If you cannot update the graphics driver on the Nano, are you able to instead run the RealSense SDK's **RealSense Viewer** tool (realsense-viewer) to test that the graphics are working correctly, please?

MAKER-park, 2021-12-16:
[image]

@MartyG-RealSense  viewer tool working is fine! so i think sdk run is good but python package have some problem... 

I can see the rgb camera image with only yuyv color setting, can this affect the python code?

MartyG-RealSense, 2021-12-16:
A D415 should support the color formats listed in the image below, with **RGB8** used as the default format.

[image]

opencv_viewer_example.py tries to use the **bgr8** format for color.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py#L36

MAKER-park, 2021-12-16:
Thank you so much for your reply!

I modified it as follows so that I can see that part and change it to yuyv.

```
if device_product_line == 'L500':
    config.enable_stream(rs.stream.color, 960, 540, rs.format.yuyv, 30) #bgr8, 30)
else:
    config.enable_stream(rs.stream.color, 640, 480, rs.format.yuyv, 30) #bgr8, 30)
```
Result after editing
[image]

A variable error occurred as follows:
The format seems to be different, but I think there are other things that I need to modify or consider when using yuyv.

I don't know which part of the code needs to be modified further.

Can I get some reference or advice?

MartyG-RealSense, 2021-12-16:
My understanding is that if you want to use yuyv as the color format then a complex conversion is required, as shown in https://github.com/IntelRealSense/librealsense/issues/4646#issuecomment-750366311

The important question is why your RealSense installation is only letting you use yuyv format for color instead of the full list of D415 color formats in the Viewer image that I provided.

MAKER-park, 2021-12-16:
okay. Try reinstalling rgb8 to allow it, or if you use opencv in a complicated way, you need to change the code. Let's approach it in different ways. 
thank you!

MAKER-park, 2021-12-17:
i'm slove this problem!

`./buildLibrealsense.sh`

i can use rgb8 format after using this shell file 
[image]

then i test opencv test is good! 

thank you for your advice!

MartyG-RealSense, 2021-12-17:
Excellent news that you found a solution, @MAKER-park - thanks very much for the update!

closed 2021-12-24 by MartyG-RealSense