realsenseai/librealsense #10071: Install realsense viewer on raspberry pi OS for desktop

EpiX-EX, 2021-12-16:
Hello,

I'm very new to Linux and to raspberry pi. I'm using VirtualBox where I have installed raspberry Pi OS for desktop in order to simulate the behavior of a raspberry pi 4. 

I'm trying to install realsense viewer so I have followed this tutorial : https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md.

I didn't execute the commands from the section "add swap" because I'm using a PC which doesn't lack space.

However I encountered issues when I have tried to execute :
```
$ sudo apt-get install -y libdrm-amdgpu1 libdrm-amdgpu1-dbgsym libdrm-dev libdrm-exynos1 libdrm-exynos1-dbgsym libdrm-freedreno1 libdrm-freedreno1-dbgsym libdrm-nouveau2 libdrm-nouveau2-dbgsym libdrm-omap1 libdrm-omap1-dbgsym libdrm-radeon1 libdrm-radeon1-dbgsym libdrm-tegra0 libdrm-tegra0-dbgsym libdrm2 libdrm2-dbgsym

$ sudo apt-get install -y libglu1-mesa libglu1-mesa-dev glusterfs-common libglu1-mesa libglu1-mesa-dev libglui-dev libglui2c2

$ sudo apt-get install -y libglu1-mesa libglu1-mesa-dev mesa-utils mesa-utils-extra xorg-dev libgtk-3-dev libusb-1.0-0-dev
```

The consol gives me : 
```

pi@raspberry:~ $ sudo apt-get install -y libdrm-amdgpu1 libdrm-amdgpu1-dbgsym libdrm-dev libdrm-exynos1 libdrm-exynos1-dbgsym libdrm-freedreno1 libdrm-freedreno1-dbgsym libdrm-nouveau2 libdrm-nouveau2-dbgsym libdrm-omap1 libdrm-omap1-dbgsym libdrm-radeon1 libdrm-radeon1-dbgsym libdrm-tegra0 libdrm-tegra0-dbgsym libdrm2 libdrm2-dbgsym
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libdrm-amdgpu1-dbgsym
E: Unable to locate package libdrm-exynos1
E: Unable to locate package libdrm-exynos1-dbgsym
E: Unable to locate package libdrm-freedreno1
E: Unable to locate package libdrm-freedreno1-dbgsym
E: Unable to locate package libdrm-nouveau2-dbgsym
E: Unable to locate package libdrm-omap1
E: Unable to locate package libdrm-omap1-dbgsym
E: Unable to locate package libdrm-radeon1-dbgsym
E: Unable to locate package libdrm-tegra0
E: Unable to locate package libdrm-tegra0-dbgsym
E: Unable to locate package libdrm2-dbgsym
```

Thanks for you help

MartyG-RealSense, 2021-12-16:
Hi @EpiX-EX  If you are using a Virtual Machine (VM), Intel recommend using VMWare Workstation Player instead of VirtualBox in the opening paragraph of the link below in order to have emulation of the USB 3 controller.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

My understanding is that VMWare Workstation Pro should also provide the necessary USB 3 emulation too.

EpiX-EX, 2021-12-17:
Hi @MartyG-RealSense,
Thanks for your answer.
I have read the link you have sent me. 
However, I'm still wondering if **it's possible to install realsense viewer on VirtualBo**x ?
I actually don't have the intel camera and I'm just trying to install the software first without expecting to go further. 
I already had VirtualBox installed on my PC and it would be easier if I could continue using this software instead of **VMware Workstation Player.**

MartyG-RealSense, 2021-12-17:
You do not need an attached camera to install the RealSense SDK (librealsense) and launch the RealSense Viewer but will be limited to testing the Viewer with pre-recorded **bag** files, which are like video recordings that contain camera data and can be interacted with in real-time as though a live camera is being used.  Sample bag files that can be tried with the Viewer to test it are provided in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

In regard to a VirtualBox VM installation though, I am not aware of a previous case where librealsense has been successfully used on VirtualBox without a camera attached though (it has failed on VirtualBox **with** a camera attached).  So I cannot provide assurances about whether a camera-less installation on VirtualBox would work, unfortunately.

closed 2021-12-30 by MartyG-RealSense