realsenseai/librealsense #10072: RealSense error calling rs2_create_device(info_list:0x55aeca582de0, index:3):     get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device

richardrl, 2021-12-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 } | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version | Ubuntu 18  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  C++ and Python | 
| Segment			|  Robotic manipulation | 

### Issue Description

Hello,
I am streaming realsense over TCP/IP using a similar to code: 

https://github.com/andyzeng/visual-pushing-grasping/blob/master/calibrate.py

However, after 15 minutes of usage, my TCP .recv call starts hanging, with the error message:

RealSense error calling rs2_create_device(info_list:0x55aeca582de0, index:3):
    get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device

What is the cause of this error? Why does it always happen with index 3?

I tried switching out the usb cable of the last camera (index 3, since I have 4 cameras) but this error keeps happening.

MartyG-RealSense, 2021-12-17:
Hi @richardrl  My understanding is that info_list is the list of camera devices that have been detected as connected, and - as you say - the index refers to the index number of the camera.  My understanding is also that the index begins at 0, so index 3 would be the 4th camera of 4 (the last camera, as you also mention).

Given that the 4 cameras can initially work correctly but one camera fails after a predictable length of time has passed, a good starting point for investigating this case would be to check whether your application has a **memory leak**.  This is where a computer's available memory is progressively consumed over a period of time until an application becomes unstable and may eventually crash.

You could check the memory usage over time using an Ubuntu system monitoring tool such as **htop**

MartyG-RealSense, 2021-12-30:
Case closed due to no further comments received.

richardrl, 2022-01-01:
```
Failed to establish dbus connectionRealSense error calling rs2_create_device(info_list:0x55a3bec0e660, index:2):
    get_xu(ctrl=1) failed! Last Error: Input/output error

```

I am getting this error now MartyG. Here is my latest full code: https://gist.github.com/richardrl/0efea5733a910ca2ae78990d1b7429fd

I don't think it's memory leak. I tried valgrind but couldn't understand what the output was. The GUI froze when I tried to run it with valgrind.

richardrl, 2022-01-01:
Please reopen, the issue is not resolved and the camera breaks down every couple minutes. I fixed the original error by not creating new devices:

I deleted this line:

`            rs2::device dev = devices[device_idx];`

and just used "devices[device_idx]" throughout my code.

MartyG-RealSense, 2022-01-01:
Valgrind is known to have problems when used to debug librealsense - as described in https://github.com/IntelRealSense/librealsense/issues/3433 - and is listed in the Known Issues section (currently unresolved problems) of the librealsense release notes for this reason.

The information provided in the box at the top of this case states that you are using kernel **4.14.13**.  Can you confirm that you have this kernel version, please?  If you are using this kernel, librealsense does not have official support for 4.14.  It skipped from 4.13 to 4.15.  Whilst librealsense can work with unsupported kernels, there is a greater risk of unpredictable consequences in regards to stability.

[image]

If you are not able to change kernel version in your project, you can instead build librealsense from source code with CMake using the **RSUSB backend** installation method.  This method is not dependent on Linux versions or kernel versions and does not require kernel patching.  Instructions for performing an RSUSB build can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692 if you have not used this method already.

MartyG-RealSense, 2022-01-07:
Hi @richardrl  Do you have an update about this case that you can provide, please?  Thanks!

richardrl, 2022-01-08:
Hi @MartyG-RealSense ,
The kernel version is: 4.15.0-166-generic

I updated the original post.

richardrl, 2022-01-08:
Would you be able to see from my code what function is failing, exactly? I'm not sure I can find where "rs2_create_device" is being called.

root@improbable009:~/realsense_docker/tests# RealSense error calling rs2_create_device(info_list:0x55fd7191d840, index:1):
    device is no longer connected! Last Error: No such file or directory

I assume, it is line 323:

        rs2::device dev = devices[device_idx];

I have opened up the repo with a Docker file. If you are able to check the script, that would be great: https://github.com/richardrl/realsense_docker

You just have to run the following, after installing Docker:

```
cd docker
./run_image.bash
./visualize_access.bash
cd ../realsense
./realsense_multicam &
```

richardrl, 2022-01-08:
I'm not sure if this was relevant, but checking htop, there seem to be many more copies of realsense_multicam process than expected...

[image]

MartyG-RealSense, 2022-01-09:
My knowledge of Docker is limited, so I am unable to debug your Docker repo unfortunately.  In librealsense release **2.50.0** Intel published a Docker tutorial, so this may be a useful reference for comparing with your own procedure.

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker

closed 2022-01-21 by MartyG-RealSense