realsenseai/librealsense #10074: Could you please provide Dynamic Calibration Print Target for A4 Size Paper?

ravijo, 2021-12-17:
Hi,

The current one given at the following URL belongs to US Letter size paper:
https://dev.intelrealsense.com/docs/calibration

While printing it on A4 size paper with  "Actual Size", I notice 3-4 mm difference. Therefore could you please provide Dynamic Calibration Print Target for A4 Size Paper?

MartyG-RealSense, 2021-12-17:
Hi @ravijo  I loaded the PDF into Word, changed the page format to A4 and re-saved the PDF.  I have attached it to this message.  Could you print it and see whether it provides you with better results, please?

[print-target-fixed-width-A4.pdf](https://github.com/IntelRealSense/librealsense/files/7735646/print-target-fixed-width-A4.pdf)

ravijo, 2021-12-17:
Hi @MartyG-RealSense 

Thank you very much for the quick response. 

I tried many options today. Converting US Letter to A4 via MS Word and Acrobat. But faced a few mm difference. 

Is it possible for you to check the attached file once please because I left my workplace already? Can you confirm the dimensions, please?

MartyG-RealSense, 2021-12-17:
My PDF reader states that the modified document is 8.27x11.69 inches, which pretty much matches to A4 size.

ravijo, 2021-12-17:
Well, I did similar steps and page was shown A4 and used "Actual Size" in printing. But the output wasn't satisfactory. that 3-4 mm difference !!!

MartyG-RealSense, 2021-12-17:
The Dynamic Calibration system offers alternatives to a printed paper target.  You can install an Android application that displays a calibration target on the android device's screen to point the camera at, or use target-less calibration (though this can be less accurate than target-based calibration).

ravijo, 2021-12-17:
@MartyG-RealSense 

I will print PDF given by you above. If it doesn't work, I am gonna use the iPhone app. Though I am concerned about possible reflection on the phone screen.

Anyway, I don't wanna use target-less calibration, because my priority is all about depth accuracy.

MartyG-RealSense, 2021-12-17:
You can significantly dampen glare from reflections by purchasing a physical thin-film **linear polarization filter** product and applying it over the camera lenses on the outside of the camera.  You can find out more about affordable linear polarization filters by going to Amazon and searching for **linear polarization sheet**.  

Any thin-film polarizer that is linear should work - circular polarizers used in 3D glasses won't work.

ravijo, 2021-12-19:
@MartyG-RealSense 

> Hi @ravijo I loaded the US Letter calibration target PDF into Word, changed the page format to A4 and re-saved the PDF. I have attached it to this message. Could you print it on A4 paper and see whether it provides you with better results, please?
> 
> [print-target-fixed-width-A4.pdf](https://github.com/IntelRealSense/librealsense/files/7735646/print-target-fixed-width-A4.pdf)

I printed this file on A4 size paper. I found 1 mm difference in width (50 mm) and 2 mm at height (100 mm)

MartyG-RealSense, 2021-12-19:
It sounds like an improvement on the 3-4 mm difference that you had at the beginning of this case, @ravijo

ravijo, 2021-12-20:
> I printed this file on A4 size paper. I found 1 mm difference in width (50 mm) and 2 mm at height (100 mm)

Sorry, this is slightly wrong. The error in height is more than I said before. Please see the attached picture below:

[image]

The above picture is a print of _print-target-fixed-width-A4.pdf_

**By the way, are you sure that you get the exact dimensions when printing on a US Letter page? Can you please print one and measure the dimensions?**  If the original PDF doesn't have correct dimensions, then A4 paper will accommodate those errors too.

ravijo, 2021-12-22:
@MartyG-RealSense 

I was able to print it perfectly. I used an A3 size page to print US Letter paper.

> print-target-fixed-width (1).pdf

Please wait for some time, I can check the new file provided by you.

MartyG-RealSense, 2021-12-22:
Thanks very much, @ravijo - I look forward to your next report.  Good luck!

MartyG-RealSense, 2022-01-02:
Hi @ravijo  Do you have an update that you can provide about this case, please?  Thanks!

ravijo, 2022-01-02:
@MartyG-RealSense 

Please give me a couple of days as it is new year vacation currently.

MartyG-RealSense, 2022-01-02:
Okay, that's no problem at all, @ravijo - thanks!

ravijo, 2022-01-04:
@MartyG-RealSense 

> I tried a different approach. I copied the calibration image from the original US Letter PDF in Word, reformatted the page size in Word to A4 and **then** pasted in the original US Letter document's image. This should mean that the original image is used within the confines of the changed page margins.
> 
> [print-target-fixed-width (1).pdf](https://github.com/IntelRealSense/librealsense/files/7745527/print-target-fixed-width.1.pdf)
 

Thank you for waiting. I printed the above PDF in A4 size paper and unfortunately, it doesn't look good to me. Please see attached picture below.

[image]

MartyG-RealSense, 2022-01-04:
Instead of trying to adapt the original US Letter PDF to A4, an alternative approach would be to buy a pack of US Letter paper and adjust the border sliders in the printer tray to fit the US Letter size, then print off the US Letter PDF.  You could locate retailer sources in your country to purchase such paper by googling for 'buy us letter paper in <name of country>'.  For example, **buy us letter paper in uk** or **buy us letter paper in australia**

ravijo, 2022-01-05:
@MartyG-RealSense 

Thank you very much. As I said I could print it.

> I was able to print it perfectly. I used an A3 size page to print US Letter paper.

At this point in time, may I request you to create a feature request (internally for the dev team), so that we can have the calibration target in A4 size too? It is not urgent but defiantly will help!

MartyG-RealSense, 2022-01-05:
As your request to create an A4 version of the calibrartion target PDF document is of a documentation nature rather than engineering or SDK programming, I will place a **Documentation** tag on this case to highlight it to the RealSense team for future attention.  This case should now be kept open, but you do not need to take further action in regard to the request.  Thanks!

Vishnu-AP, 2025-02-20:
Hi There @MartyG-RealSense, Any update on the A4 sized pdf document with the target?

MartyG-RealSense, 2025-02-20:
Hi @Vishnu-AP  An A4 version of the calibration target was not created.  I made another attempt today at creating an A4 printable PDF of the target myself, which can be downloaded from the link below.

[A4-Calibration.pdf](https://github.com/user-attachments/files/18885124/A4-Calibration.pdf)

Vishnu-AP, 2025-02-20:
I tried with this A4 version and it seems to work for me, I have about 1mm of error only in the square boxes below which should be good. I have a few questions with respect to the calibration though, it would be great if you could help me out
1. I used the sheet printed to calibrate and it said the calibration results are now saved to the camera, so I went on to check with the depth quality tool and these are the values that I am getting:
Z accuracy: 0.04%
Plane Fit RMS Error: 0.16%
Fill Rate: 100%
Subpixel RMS error: 0.03 Pixels
The angle mentioned is 3.20 degrees
Are these values good?
2. If these values are good, how do I get the intrinsic values now? will they be updated to the realsense-viewer application?

MartyG-RealSense, 2025-02-20:
Yes, those are good values.  

You can list the intrinsic values for each stream format using the RealSense SDK's **rs-enumerate-devices** tool by launching it in calibration information mode with the command `rs-enumerate-devices -c`

Vishnu-AP, 2025-02-21:
Thanks for the info!
I tried with rs-enumerate-devices -c

Everything looks good except for this
 21/02 10:25:47,367 ERROR [139314910298688] (rs.cpp:237) [rs2_get_video_stream_intrinsics( from:0x578c923caba0, intr:[ -1237344144x32766  p[0 0]  f[0 0]  UNKNOWN [4.59149e-41 -2.85547e-06 4.59149e-41 0 0] ] ) Not Implemented] No intrinsics are available for this stream profile!
 21/02 10:25:47,367 ERROR [139314910298688] (rs.cpp:237) [rs2_get_video_stream_intrinsics( from:0x578c923ba7f0, intr:[ -1237344144x32766  p[0 0]  f[0 0]  UNKNOWN [4.59149e-41 -2.85547e-06 4.59149e-41 0 0] ] ) Not Implemented] No intrinsics are available for this stream profile!
 21/02 10:25:47,368 ERROR [139314910298688] (rs.cpp:237) [rs2_get_video_stream_intrinsics( from:0x578c923c8300, intr:[ 640x480  p[320.459 240.305]  f[389.323 389.323]  Brown Conrady [0 0 0 0 0] ] ) Not Implemented] No intrinsics are available for this stream profile!
 21/02 10:25:47,368 ERROR [139314910298688] (rs.cpp:237) [rs2_get_video_stream_intrinsics( from:0x578c923c4030, intr:[ 640x480  p[320.459 240.305]  f[389.323 389.323]  Brown Conrady [0 0 0 0 0] ] ) Not Implemented] No intrinsics are available for this stream profile!
 21/02 10:25:47,368 ERROR [139314910298688] (rs.cpp:237) [rs2_get_video_stream_intrinsics( from:0x578c923caca0, intr:[ 424x240  p[212.254 120.179]  f[214.939 214.939]  Brown Conrady [0 0 0 0 0] ] ) Not Implemented] No intrinsics are available for this stream profile!
 21/02 10:25:47,368 ERROR [139314910298688] (rs.cpp:237) [rs2_get_video_stream_intrinsics( from:0x578c923c22b0, intr:[ 424x240  p[212.254 120.179]  f[214.939 214.939]  Brown Conrady [0 0 0 0 0] ] ) Not Implemented] No intrinsics are available for this stream profile!
 21/02 10:25:47,368 ERROR [139314910298688] (rs.cpp:237) [rs2_get_video_stream_intrinsics( from:0x578c923c83b0, intr:[ 640x480  p[320.459 240.305]  f[389.323 389.323]  Brown Conrady [0 0 0 0 0] ] ) Not Implemented] No intrinsics are available for this stream profile!
 21/02 10:25:47,368 ERROR [139314910298688] (rs.cpp:237) [rs2_get_video_stream_intrinsics( from:0x578c923c4180, intr:[ 640x480  p[320.459 240.305]  f[389.323 389.323]  Brown Conrady [0 0 0 0 0] ] ) Not Implemented] No intrinsics are available for this stream profile!
Intrinsic Parameters:

I am not sure if this is anything to worry about

MartyG-RealSense, 2025-02-21:
@Vishnu-AP  On certain RealSense 400 Series camera models (D415, D435 and D435i) it is normal for all distortion coefficients to be zero.

Regarding the intrinsics errors, I would recommend resetting your camera to its factory-new default calibration in the RealSense Viewer tool using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 in case the errors are being caused by a corrupted 'calibration table' inside the camera hardware (a table that the calibration values are stored in).  Performing the factory-reset procedure creates a new calibration table in place of a corrupted one.

MartyG-RealSense, 2025-02-21:
The sets of intrinsic values for the different resolutions that are printed after the error messages look fine, so if it is providing the information that you need then you can ignore the errors.

Yes, resetting the calibration to factory-new would overwrite your camera's current calibration.

Vishnu-AP, 2025-02-21:
Okay got it! sorry for the confusion. I also wanted to know if there are any other distortion models that an be printed, currently it is the Brown Conrady model, can I get something like the Rad Tan model?

MartyG-RealSense, 2025-02-21:
The distortion model is applied inside the camera hardware before the data is sent through the USB cable to the computer, so the model cannot be changed.

If the RealSense SDK's **software-device** 'virtual camera' API was used - which allows the definition of a simulated camera without having a real one attached - then you could define your own custom distortion.

https://github.com/IntelRealSense/librealsense/tree/e1688cc318457f7dd57abcdbedd3398062db3009/examples/software-device

Vishnu-AP, 2025-02-24:
Okay great! This pretty much sorts my issue
Thanks for the help @MartyG-RealSense !

MartyG-RealSense, 2025-02-24:
You are very welcome, @Vishnu-AP - I'm pleased that I could help!

MartyG-RealSense, 2026-10-23:
This open GitHub issue is being closed due to its significant age and no plans to implement the suggested creation of an A4 version of the calibration print target.

closed 2026-10-23 by MartyG-RealSense