realsenseai/librealsense #10075: Where is CustomCalibration tool and Chessboard Pattern?

ravijo, 2021-12-17:
Hi,

Based on [Intel RealSense D400 Series Custom Calibration Whitepaper](https://dev.intelrealsense.com/docs/d400-series-custom-calibration-white-paper), I should be able to find CustomCalibration tool. But I can't.

I am sorry but can you please provide:

- CustomCalibration
- Chessboard Pattern

MartyG-RealSense, 2021-12-17:
Hi @ravijo  You have already located the official chessboard calibration target pattern, as mentioned in your other case at https://github.com/IntelRealSense/librealsense/issues/10074

In regard to the Calibration API for developing a custom calibration tool, you can visit the link below to find the PDF document of the **Programmer's guide for Intel RealSense D400 Series calibration tools and API**.  The current edition at the time of writing this is dated July 2021.

https://dev.intelrealsense.com/docs/d400-dynamic-calibration-quick-start-guide

My understanding is that the Calibration API is obtained by installing the Calibration Tool, as described on page 11 of the above PDF Programmer's Guide.

[image]

You do not need to necessarily develop your own custom calibration application though, as the standard Dynamic Calibration pre-made tool should be sufficient for 99% of RealSense users.

There is also the alternative option of calibrating the camera using the **On-Chip** calibration tool, which is accessible under the 'More' option of the RealSense Viewer's options side-panel or with program scripting.  On-Chip calibration has the advantage of being built into the camera firmware driver and so does not require installation of a separate software tool like the Dynamic Calibration tool does.

ravijo, 2021-12-17:
Hello @MartyG-RealSense 

I understand. 

Just to confirm, do you mean that the CustomCalibration tool is not required because of the following two tools

- Dynamic Calibration tool (for extrinsic calibration)
- On-Chip Calibration (for intrinsic calibration)

Please see page number 22 of Intel-RealSense-D400-Series-Custom-Calibration-Whitepaper.pdf. If I understood you correctly, the picture/process shown on page number 22 is not required.

Please confirm.

MartyG-RealSense, 2021-12-17:
You are correct, you do not need to create your own custom calibration application and the pre-made Dynamic Calibration or On-Chip tools should be fine.  In the 4 year history (so far) of the 400 Series cameras, there have only been a few enquiries that I know of regarding custom-developing a calibration tool with the Calibration API.

ravijo, 2021-12-17:
Thanks a lot.

closed 2021-12-17 by ravijo