realsenseai/librealsense #10076: Using and generating mex files- MATLAB wrapper support

Shoghair, 2021-12-17:
Hello, 

We have been trying to use the RealSense SDK on MATLAB by creating an S-Function using the mex files present in the MATLAB wrapper wen installed using the windows installer. However, Simulink always returns an error ` S-Function libreasense_mex does not exist, although in path`. I have tried to use the [C++ code in the wrapper ](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/wrappers/matlab/librealsense_mex.cpp) to generate a new mex file to use but am having issues as I obtain a large amount of errors, some of which shown below:

`                                                          ^~~~~~~~~~~~~~
In file included from C:\Users\Silicon
Synapse\Downloads\librealsense-master\librealsense-master\wrappers\matlab\librealsense_mex.cpp:1:0:
C:\Users\Silicon Synapse\Downloads\librealsense-master\librealsense-master\wrappers\matlab\MatlabParamParser.h:251:92: error:
specialization of 'template<class T> static mxArray* MatlabParamParser::mx_wrapper_fns<T, typename
std::enable_if<((std::integral_constant<bool, ((std::is_arithmetic<_Tp>::value || std::is_pointer<_Tp>::value) ||
std::is_enum<_Tp>::value)>::value && (! std::integral_constant<bool, (std::is_base_of<rs2::frame, T>::value ||
std::is_base_of<rs2::stream_profile, T>::value)>::value)) && (! is_array_type<T>::value))>::type>::wrap(T&&)' in different namespace
[-fpermissive]
 template<> mxArray* MatlabParamParser::mx_wrapper_fns<const char *>::wrap(const char*&& str)
                                                                                            ^
C:\Users\Silicon Synapse\Downloads\librealsense-master\librealsense-master\wrappers\matlab\MatlabParamParser.h:192:21: error:   from
definition of 'template<class T> static mxArray* MatlabParamParser::mx_wrapper_fns<T, typename
std::enable_if<((std::integral_constant<bool, ((std::is_arithmetic<_Tp>::value || std::is_pointer<_Tp>::value) ||
std::is_enum<_Tp>::value)>::value && (! std::integral_constant<bool, (std::is_base_of<rs2::frame, T>::value ||
std::is_base_of<rs2::stream_profile, T>::value)>::value)) && (! is_array_type<T>::value))>::type>::wrap(T&&)' [-fpermissive]
     static mxArray* wrap(T&& var)
`

Any possible help/guidance would be much appreciated.

MartyG-RealSense, 2021-12-18:
Hi @Shoghair  This is the first question that has been posted on the RealSense forums about adapting the RealSense MATLAB wrapper to include an S-function, so thanks in advance for your patience.  :)

As a starting point in investigating this case, a member of staff on the MATLAB support forum who was answering an S-Function ... does not exist' question suggested compiling the S-Function into a **mexw64** using the "mex" command.

https://www.mathworks.com/matlabcentral/answers/735042-s-function-does-not-exist-but-is-clearly-in-the-matlab-path

Shoghair, 2021-12-19:
Hi @MartyG-RealSense . Thank you for the response! I have been trying to compile using the mex command. 
My setup is as follows:
`>> mex -setup C++
MEX configured to use 'MinGW64 Compiler (C++)' for C++ language compilation.`
And I compile using the following command. 
`>> mex librealsense_mex.cpp`
I keep getting multiple errors like the ones attached in my initial question which appear to be a compilation issue? Mainly from [MatlabParamParser.h](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/wrappers/matlab/MatlabParamParser.h)
Should I be generating the mex from additional/multiple cpp files or just the[ librealsesne_mex.cpp](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/wrappers/matlab/librealsense_mex.cpp)?

I have also followed the recommendation reference https://www.mathworks.com/matlabcentral/answers/735042-s-function-does-not-exist-but-is-clearly-in-the-matlab-path. I have my .cpp file along with the already generated mexw64 in the matlab wrapper when downloaded using windows installer in the same path.

MartyG-RealSense, 2021-12-19:
I researched your problem further but there is no other reference material available in regard to using S-Functions with librealsense that I can find, unfortunately.  The link below suggests further things to check.

https://stackoverflow.com/questions/8953759/s-function-in-simulink-matlab

closed 2022-01-01 by MartyG-RealSense