realsenseai/librealsense #10077: Error 403 from "librealsense.intel.com" on Ubuntu 21

CyberFoxHax, 2021-12-18:
I followed these instructions: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
Had success on a testbench running an older Ubuntu but not on my Ubuntu 21.10 workstation.

terminal dump:
```
yuqi@yuqi:~$ export http_proxy="http://127.0.0.1:20171"
yuqi@yuqi:~$ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
[sudo] password for yuqi: 
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.QtBL9KP8d6/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
yuqi@yuqi:~$ sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
Repository: 'deb https://librealsense.intel.com/Debian/apt-repo impish main'
Description:
Archive for codename: impish components: main
More info: https://librealsense.intel.com/Debian/apt-repo
Adding repository.
Press [ENTER] to continue or Ctrl-c to cancel.
Adding deb entry to /etc/apt/sources.list.d/archive_uri-https_librealsense_intel_com_debian_apt-repo-impish.list
Adding disabled deb-src entry to /etc/apt/sources.list.d/archive_uri-https_librealsense_intel_com_debian_apt-repo-impish.list
Hit:1 http://mirrors.aliyun.com/ubuntu focal InRelease
Get:2 http://mirrors.aliyun.com/ubuntu focal-security InRelease [114 kB]
Hit:3 http://mirrors.aliyun.com/ubuntu focal-updates InRelease                 
Hit:4 https://mirrors.cloud.tencent.com/ubuntu impish InRelease                
Hit:5 http://mirrors.aliyun.com/ubuntu focal-proposed InRelease                
Get:6 https://mirrors.cloud.tencent.com/ubuntu impish-updates InRelease [110 kB]
Hit:7 http://mirrors.aliyun.com/ubuntu focal-backports InRelease               
Get:8 http://mirrors.aliyun.com/ubuntu focal-security/main amd64 DEP-11 Metadata [35,7 kB]
Get:9 http://mirrors.aliyun.com/ubuntu focal-security/universe amd64 DEP-11 Metadata [66,1 kB]
Get:10 http://mirrors.aliyun.com/ubuntu focal-security/multiverse amd64 DEP-11 Metadata [2.464 B]
Hit:11 https://download.mono-project.com/repo/ubuntu vs-focal InRelease        
Get:12 https://mirrors.cloud.tencent.com/ubuntu impish-backports InRelease [101 kB]
Get:13 https://mirrors.cloud.tencent.com/ubuntu impish-security InRelease [110 kB]
Hit:14 https://packages.microsoft.com/ubuntu/20.04/prod focal InRelease        
Err:15 https://librealsense.intel.com/Debian/apt-repo impish InRelease         
  403  Forbidden [IP: 13.227.66.47 443]
Reading package lists... Done
E: Failed to fetch https://librealsense.intel.com/Debian/apt-repo/dists/impish/InRelease  403  Forbidden [IP: 13.227.66.47 443]
E: The repository 'https://librealsense.intel.com/Debian/apt-repo impish InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
yuqi@yuqi:~$ sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package librealsense2-dkms is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'librealsense2-dkms' has no installation candidate
yuqi@yuqi:~$ sudo apt-get install librealsense2-utils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package librealsense2-utils
yuqi@yuqi:~$ 
```

I tried with and without proxy, same result. (In China if it makes a difference)

```
yuqi@yuqi:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 21.10
Release:	21.10
Codename:	impish
yuqi@yuqi:~$ cat /proc/version
Linux version 5.13.0-22-generic (buildd@lgw01-amd64-012) (gcc (Ubuntu 11.2.0-7ubuntu2) 11.2.0, GNU ld (GNU Binutils for Ubuntu) 2.37) #22-Ubuntu SMP Fri Nov 5 13:21:36 UTC 2021
```

MartyG-RealSense, 2021-12-18:
Hi @CyberFoxHax  The librealsense SDK currently supports up to Ubuntu 20.04 (Focal).  You are the first RealSense user who has reported trying to use it with Ubuntu 21.  

If you build librealsense from source code with CMake using the **RSUSB backend** method instead of building with packages then it will not be dependent on Linux versions or kernel versions and will not require patching.  https://github.com/IntelRealSense/librealsense/issues/6689#issuecomment-648758973 has instructions for performing an RSUSB backend installation.

MartyG-RealSense, 2021-12-30:
Case closed due to no further comments received.

closed 2021-12-30 by MartyG-RealSense

CyberFoxHax, 2021-12-30:
Busy period. Set up a Windows environment for Realsense. This is a temporary workaround, i'll try to build from source in a few weeks.

MartyG-RealSense, 2021-12-30:
Okay, thanks very much @CyberFoxHax for the update.  I look forward to your next report.  Good luck!

szymonborkowski, 2023-11-11:
Hello @MartyG-RealSense, I came across this thread as I'm trying to install the SDK on Linux Mint which is running Ubuntu v21.2. I noticed the SDK only runs on v20 and v22, is there a reason why v21 got skipped?

MartyG-RealSense, 2023-11-11:
Hi @szymonborkowski  I am not aware of the development reasons for going from 20.04 to 22.04 but there has not been any demand from the RealSense community for 21.2 support. 

You should be able to use the SDK with 21.2 if you build it from source code with CMake with the build flag **-DFORCE_RSUSB_BACKEND=TRUE** included in the CMake build instruction though.  This is because an RSUSB build of the SDK bypasses the kernel and so it is not dependent on Linux versions or kernel versions and does not need to be kernel patched. 
 Instructions for an RSUSB build can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692