realsenseai/librealsense #10080: rs-server not work

yuxue521, 2021-12-19:
I want to use a network connection D430 to use RS-server
 test librealsense-2.49.0 and librealsense-2.47.0 

on Ubuntu  build
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DCMAKE_INSTALL_PREFIX=../Cinstall -DBUILD_NETWORK_DEVICE=ON

run  rs-server 
./rs-server 
Rs-server is running

usb realsense-viewer  add network device ip 127.0.0.1
Received error
[getStreams] error:404 Stream Not Found 404

RealSenseSupport, 2021-12-20:
@yuxue521 Did you follow the steps in https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-6-setting-static-ip ？

MartyG-RealSense, 2021-12-20:
@RealSenseSupport  Thanks very much for your assistance with the question by @yuxue521   :)

yuxue521, 2021-12-20:
@RealSenseSupport  I don't have PI4 I just want to test the network under Ubuntu. I tried local IP (127.0.0.1) and 192.168.31.200 failed
[image]

MartyG-RealSense, 2021-12-26:
Hi @yuxue521  Are you still experiencing this problem and require further assistance, please?  Thanks!

yuxue521, 2021-12-26:
@MartyG-RealSense I need, i haven't succeeded

MartyG-RealSense, 2021-12-26:
I am not familiar with the IP address connection method that you used: **usb realsense-viewer add network device ip 127.0.0.1**

I note that you are not using a Raspberry Pi 4.  

Is rs-server installed on a non-Pi remote computer that the camera is attached to, and you are connecting to that computer with ssh from a central host computer?

yuxue521, 2021-12-26:
@MartyG-RealSense Both rsserver and realsense-viewer work on the local PC, I was to test the rsserver network, If it succeeds, other devices should be able to connect.

MartyG-RealSense, 2021-12-26:
**rs-server** is intended to be built on a second Ubuntu-equipped computer / computing device so that a camera attached to that computer can be communicated with over a network connection from a main computer (the central 'host') that has realsense-viewer installed on it.   

For example, the open-source ethernet networking white-paper describes a setup where there is a central host PC and a remote computer (Pi 4) that has the camera attached to it and rs-server installed on it.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

[image]

If only one computer (the local PC) was being used then there would be no need for setting up the camera as a Network Camera, as the camera could be accessed by realsense-viewer directly thorugh the USB cable.

Perhaps I am misunderstanding something about your one-computer networking project.  If I am then please provide more information.  Thanks!

yuxue521, 2021-12-27:
For the convenience of testing, I use my local PC for testing. Why doesn’t this work? I tried to use two ubuntu PC and the test also failed.
I think his working principle should be to obtain video data from USB and forward it to the network (rtsp/rtp) through LIVE555. For the network, it should not be distinguished between the local machine and other PC. Isn't that the case?

MartyG-RealSense, 2021-12-28:
I see that at the top of this case the camera model is listed as **D430**.  Is that correct please?  If it is then you would not be able to stream RGB video from a D430 as that model (a caseless Depth Module Kit) does not contain an RGB sensor.

yuxue521, 2021-12-28:
@MartyG-RealSense Yes, I use D430, how do I use Z16 to RGB or Y8 / Y16 to RGB, have a document?

MartyG-RealSense, 2021-12-28:
The D430 does not have RGB available from the left infrared sensor either (the D415 and D450 / D455 models do).  I therefore cannot see any way for you to obtain RGB unless you use a second camera to provide the RGB source and sync it with the RealSense camera.

You could obtain a monochrome RGB-like image from the left infrared sensor (Y8) if you minimized the Laser Power setting or turned the projector off in order to make the IR dot pattern disappear from the image.  It would not be a true RGB format image though.

yuxue521, 2021-12-28:
Thank you, I understand

closed 2022-01-02 by yuxue521