realsenseai/librealsense #10081: D415/D435 Real Sense Face Authentication F455 Functionality

luisruizlatic, 2021-12-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> ??) | 
| Operating System & Version |   {Win 10}  | 
| Kernel Version (Linux Only)    |  N/A | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.0.0 }   | 
| Language                            |  { C# }  | 
| Segment			|  { others }   | 

### Issue Description
Hello, We integrated the camera F455 a month ago to our POS application and it works great!.

Sadly we saw the discontinuation notice in your web page, so we got the camera D415/D435 but this SDK doesn't have the same functionality.

Do you know what third party library can we use for the camera D415/D435 to make it work in the same way than the F455? 

Are you planning to add the Realsense ID functionality to this SDK ?

Thanks in advance.

MartyG-RealSense, 2021-12-20:
Hi @luisruizlatic  If you require a commercial-grade pre made face authentication solution then you could investigate **LIPSFace**, a RealSense-powered product by LIPS Corp, a manufacturer of a range of products that incorporate RealSense camera technology.  

https://www.lips-hci.com/lipsface

Please bear in mind that with RealSense-powered products produced by non-Intel companies, technical support is provided by those manufacturers and not Intel.

If the Intel RealSense ID F455 works well for your product then stock of that product is still available for purchase, knowing in advance of course that it will not be available for further stock purchase in future if you need to order additional units later.

In regard to 400 Series cameras, it is possible to develop your own applications to perform facial authentication and anti-spoofing with 400 Series cameras though.  The RealSense SDK provides a compatibility wrapper for the popular **Dlib** face library.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/dlib

I am not aware of future plans to add RealSense ID functionality to the RealSense SDK.  The End Of Life notice for the F450 / F455 states that the recommended alternative will be to use a 400 Series camera with third-party face authentication software.

For a commercial software solution for facial authentication, you could consider 3DiVi's **Face SDK**. 

**Face SDK**
https://face.3divi.com/

**RealSense documentation for Face SDK**
https://docs.facesdk.3divi.com/docs/3.10.0/tutorials/cpp/liveness_detection

MartyG-RealSense, 2022-01-02:
Case closed due to no further comments received.

closed 2022-01-02 by MartyG-RealSense

luisruizlatic, 2022-01-11:
Thanks for your help MartyG, unfortunately any of this solutions are compared to the F455.