realsenseai/librealsense #10084: How to set the timeout period for acquiring image data

shiguangzqz, 2021-12-21:
[image]
I want to reduce timeout，which api can i use to reduce the frame arrive timeout？

MartyG-RealSense, 2021-12-21:
Hi @shiguangzqz  If the program has not been receiving frames for long enough for a **Frame didn't arrive within 15000** message to be generated then something has likely gone wrong, such as a freeze-up in the program's operation.  In past cases, this message tends to occur when closing the pipeline with an instruction such as **pipe.Stop()**.  

A good reference about freeze after pipeline stop is https://github.com/IntelRealSense/librealsense/issues/7252 - in that case, it was suggested in https://github.com/IntelRealSense/librealsense/issues/7252#issuecomment-695194103 that stopping with **sensor.Stop()** instead of pipe.Stop() may help.  Other RealSense users found that simply updating their SDK and firmware driver versions resolved this error.

shiguangzqz, 2021-12-21:
> Hi @shiguangzqz If the program has not been receiving frames for long enough for a **Frame didn't arrive within 15000** message to be generated then something has likely gone wrong, such as a freeze-up in the program's operation. In past cases, this message tends to occur when closing the pipeline with an instruction such as **pipe.Stop()**.
> 
> A good reference about freeze after pipeline stop is #7252 - in that case, it was suggested in [#7252 (comment)](https://github.com/IntelRealSense/librealsense/issues/7252#issuecomment-695194103) that stopping with **sensor.Stop()** instead of pipe.Stop() may help. Other RealSense users found that simply updating their SDK and firmware driver versions resolved this error.

I will encounter this situation when I unplug the camera usb when acquiring image data
My program timeout is 10s, I hope this timeout error can be returned in 5s or 8s

MartyG-RealSense, 2021-12-21:
This sounds like a similar situation to one that you experienced in https://github.com/IntelRealSense/librealsense/issues/6518 in June 2020 when **Frame didn't arrive within 15000** was occurring because the pipeline was apparently not being closed down cleanly.  I understand that in https://github.com/IntelRealSense/librealsense/issues/6518#issuecomment-639957507 within that case you tried dealing with the problem by catching an exception.

A faster way to deal with a problem caused by a USB 'detach' event may be to listen for a USB unplug 'detach event' with a function called **set_devices_changed_callback**

**C++**
https://github.com/IntelRealSense/librealsense/issues/931 

**Python**
https://github.com/IntelRealSense/librealsense/issues/4212

shiguangzqz, 2021-12-21:
> 

I have see the question#931，but I still unknow how to use rs2::contenxt::set_devices_changed_callback，can you show me how to use this？

shiguangzqz, 2021-12-21:
> This sounds like a similar situation to one that you experienced in #6518 in June 2020 when **Frame didn't arrive within 15000** was occurring because the pipeline was apparently not being closed down cleanly. I understand that in [#6518 (comment)](https://github.com/IntelRealSense/librealsense/issues/6518#issuecomment-639957507) within that case you tried dealing with the problem by catching an exception.
> 
> A faster way to deal with a problem caused by a USB 'detach' event may be to listen for such an event with a function called **set_devices_changed_callback**
> 
> **C++** #931
> 
> **Python** #4212

I have see the question#931，but I still unknow how to use rs2::contenxt::set_devices_changed_callback，can you show me how to use this？

MartyG-RealSense, 2021-12-21:
A C++ example of using set_devices_changed_callback is provided below.

```
ctx.set_devices_changed_callback([&](rs2::event_information& info)
{
if (info.was_removed(my_device))
handle_device_removale();
});
```

closed 2022-01-02 by MartyG-RealSense