realsenseai/librealsense #10085: librealsense does not use a fallback code when compiled with CUDA but no GPU is available

guillaumeautran, 2021-12-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.4.0.66                                        | 
| Platform                                 | PC|
| SDK Version                            |  2.50.0                        | 
| Language                            |  C++                         | 
| Segment			|  Robot                          | 

### Issue Description
Based on the code and actual testing, the librealsense compiled with `BUILD_WITH_CUDA` option does not fall back to a CPU implementations when a GPU is not available.
The selection of backend processing code is done using a `#ifdef` strategy in here:
https://github.com/IntelRealSense/librealsense/blob/5ff27fca5aaeec4736d6bb3bfb958fee82ee198b/src/proc/processing-blocks-factory.cpp#L13-L30
as opposed to performing a fallback if creating a CUDA context does not work.

MartyG-RealSense, 2021-12-21:
Thanks so much @guillaumeautran for your PR.  I have added an **Enhancement** tag to this case to indicate that it should be kept open whilst the PR is active.  Thanks again!

MartyG-RealSense, 2022-03-19:
Hi again @guillaumeautran  As your PR at https://github.com/IntelRealSense/librealsense/pull/10086 did not pass build checks, would you like to take a look at it to see whether it can be fixed, please?  Thanks!

MartyG-RealSense, 2026-10-25:
Case closed due to the Pull Request (PR) associated with it having been already closed.

closed 2026-10-25 by MartyG-RealSense