realsenseai/librealsense #10088: pictures transmission between two pc

flyover-26, 2021-12-22:
hi!
I want to use d435i with ros wapper on one pc-1.And use another pc-2 to receive pictures from pc-1 which have a d435i.
However,I found the speed transmission is very slow although I change resolution to 640*360,fps 15.
And in my experienment,I only need color images which is rostopic"/color/image/raw".And I found there are a lot of rostopic which may decrease the speed.
So could you tell me how to shut down these rostopic?or some methods to speed up  transmission?
Thank you a lot!

[image]

MartyG-RealSense, 2021-12-22:
Hi @flyover-26  There is a discussion at https://github.com/IntelRealSense/realsense-ros/issues/1990 about **unsubscribing** from ROS topics to reduce processing.

If you are using networking, slowdown can sometimes occur, especially if using wireless networking (WLAN / wi-fi), as described in https://github.com/IntelRealSense/realsense-ros/issues/2052

flyover-26, 2021-12-23:
THANK YOU !
I got another question is that when I test a slam project.I use d435i as an input.
Everything goes well when I use resolution 1280*720.However,when I change resolution to 640*420 or other,slam will crash with no reason.Could you give me some advice on this?thank you so much.

this is the launch  file I use to run d434i.
[test.txt](https://github.com/IntelRealSense/librealsense/files/7767230/test.txt)

flyover-26, 2021-12-23:
1920✖720 is also ok.Which means 1280*720 or above is no problem for the project but below.

flyover-26, 2021-12-23:
And I use gdb to find this error.
received signal SIGSEGV, Segmentation fault.
[image]

MartyG-RealSense, 2021-12-23:
Have you tried 640x480?

Also, if you are on a USB 2.1 connection instead of USB 3, 848x480 would not be supported by RealSense.  Also, 1280x720 would be limited to 6 FPS when used as depth resolution but be able to use up to 15 FPS when used for the RGB resolution.

flyover-26, 2021-12-23:
yes.I have tried 640x480.And I got the same error.
I am using USB 3 and librealsense v2.45.0

MartyG-RealSense, 2021-12-23:
Looking at the D435i launch file that you kindly provided, I note that you have **enable_fisheye** set to True.  This setting is False by default on the standard wrapper launch files such as rs_camera.launch.  The D435i does not have fisheye sensors as they are only on the RealSense T265 Tracking Camera model.

flyover-26, 2021-12-23:
thank you so much for your patience!
I change it to false and still get the error.
this is launch file I use now.
[launch.txt](https://github.com/IntelRealSense/librealsense/files/7769153/launch.txt)
and log file.
[image]

MartyG-RealSense, 2021-12-23:
I see that the log contains a **likely requires patch for fourcc code RW16!** warning, suggesting that librealsense might be unpatched if you built it from source code.

Let's try using rs_camera.launch with a custom 640x480 color stream configuration to determine whether the problem is in your launch file.

**roslaunch realsense2_camera rs_camera.launch color_width:=640 color_height:=480 color_fps:=30**

flyover-26, 2021-12-23:
YES.I built librealsense from source code.
I run commmand you provided and still get the warning.
[image]

MartyG-RealSense, 2021-12-23:
When you installed from source code, did you patch the kernel before building by going to the root directory of librealsense and using the command below:

**./scripts/patch-realsense-ubuntu-lts.sh**

This is in point 4 of the **Prerequisites** section of Intel's official documentation for installing librealsense from source code.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites

flyover-26, 2021-12-23:
I don't patch my computer before.I read the patch-realsense-ubuntu-lts.sh.
Will this .sh file change my version of kernel?My computer can only be used under  LTS kernel 5.4.And I noticed that this .sh file will download linux-headers-4.15.0-161 linux-headers-4.15.0-161-generic.

By the way,I want to use d435i on Xavier NX.Do I need patch its kernel with the same method you provide?Because NX get the same warning.
Thank you!

MartyG-RealSense, 2021-12-23:
It would not change the kernel version.   If you are concerned about making changes to the kernel then you can avoid doing so by building librealsense from source code with the **RSUSB backend** method.  This method is not dependent on Linux versions or kernel versions and does not require patching.  A guide for building on Jetson boards using this method is at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 and this method has been successfully tested with Xavier NX,

flyover-26, 2021-12-23:
THANK YOU SO MUCH!
When I use RSUSB backend method,I ran the command"cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=..." on my pc.

I got an error below.
[image]

MartyG-RealSense, 2021-12-23:
A Jetson user who experienced this error stated in https://github.com/IntelRealSense/librealsense/issues/6449#issuecomment-650794408 that they could build successfully by leaving out the **-DPYTHON_EXECUTABLE** command or by using **-DPYTHON_EXECUTABLE=/usr/bin/python**

If you do not require Python support to be included in your librealsense build then you could also leave out the Python commands from the CMake build instruction.

**cmake ../ -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true**

flyover-26, 2021-12-24:
hi!
I use -DPYTHON_EXECUTABLE=/usr/bin/python to solve the problem.thank you.
But when I run roslaunch realsense2_camera rs_camera.launch color_width:=640 color_height:=480 color_fps:=30
I got problem like this.
[image]
 and when I open realsense_viewer,I see some error.
[image]

MartyG-RealSense, 2021-12-24:
Do the warnings still occur if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch, please?

flyover-26, 2021-12-24:
yes.
I still get the warning.
My librealsense is v2.45.

And when I try to run this command " cmake ../ -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true"
I got an error like can not find CUDA  library.

thank you for your help.

MartyG-RealSense, 2021-12-24:
Bearing in mind that you were using 15 FPS at the beginning of this case, how does the ROS launch behave if you set FPS to 15 instead of 30?

**roslaunch realsense2_camera rs_camera.launch color_width:=640 color_height:=480 color_fps:=15**

If you are using a Jetson Xavier that has CUDA installed then it should be able to use CUDA as Jetson boards have the Nvidia graphics GPU that is a requirement of CUDA.  Can you confirm whether you tested the CMake build on a Jetson board, please?

flyover-26, 2021-12-24:
yes.I try roslaunch realsense2_camera rs_camera.launch color_width:=640 color_height:=480 color_fps:=15.
And still got the warning.
I build librealsense on xavier nx.And I use method below.

Download the zip file from https://github.com/IntelRealSense/librealsense/releases/. I am using version 2.38.1

Extract the file, cd into the extracted file .

Create a dir called build and cd into it.

Run the CMake command to test to see if the build will work.
cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release -DPYTHON_EXECUTABLE=/usr/bin/python

If you have issues with the above step, check out this, this and this

Still in the build dir. Run make -j4 and then sudo make install

By the way,after we solve patch for fourcc code RW16.My slam project can run correctly,but it is not stable enough.It will crash sometimes because of unkown problem about realsense i guess.
Thank you for your helping!

flyover-26, 2021-12-24:
I use version v2.45

MartyG-RealSense, 2021-12-24:
It's excellent to hear that you resolved the 'fourcc code RW16' issue!

Are you using ROS wrapper version **2.3.0** please?  This is the correct wrapper version to match with librealsense SDK 2.45.0, as it is important to match a particular wrapper as closely as possible with the librealsense version listed in that wrapper's release notes.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.0

flyover-26, 2021-12-24:
yes，it is matched.
And I have no idead how to solve this.
Are there any other reason to lead to this warning?
thank you

MartyG-RealSense, 2021-12-24:
How many cameras do you have attached to the same PC?  There have been past Jetson cases where if more than one camera is attached then a **Failed to set power state** error may occur, whereas if only one camera is attached then it works normally.

closed 2022-01-05 by MartyG-RealSense