realsenseai/librealsense #10089: IR Laser not visible on Slave camera with short exposure time.

PorschefanRoel, 2021-12-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430 | 
| Firmware Version                   |Tested both with 05.12.15.50 and 05.13.0.50 | 
| Operating System & Version |  Win 10 | 
| Kernel Version (Linux Only)    | -                                         | 
| Platform                                 | PC  |
| SDK Version                            |  Realsense Viewer v2.49.0                          | 
| Language                            |  {C#}                          | 

We have the following setup and seem to have a problem with the HW Sync combined with the internal laser

D435 Master (Inter Cam Sync Mode 1) <Sync Cable> D435 Slave (Inter Cam Sync Mode 2)
Both cameras are point in the same direction, with a distance of +/- 45cm between them, and the object is around 50cm away.

The following settings are set fixed:
Resolution: 848x480
FPS: 6
Exposure Time: 1000
Laser Power: 360
Gain: 16

[image]

As can be seen, the master gives a descent depth image and on the IR image can be seen that the IR Laser is working.
But the Slave D435 is messy and doesn't show any laser data on the IR image.

When I slowly increase the Exposure to 1100, 1200, 1300 etc. Then slowly a laser is becomming visible int the IR image

When the exposure is 1600 this is visible:
[image]

Can it be that the laser is slightly out of sync with the capturing of the images, or can it be any other problem?

When selecting Emitter On Off on the Slave Camera, I can see on the master IR image that the laser is actually flashing, but it is not shown in the slave IR
[image]

When selecting the option Emitter Always On, there still isnt a stable IR Light:
[image]

When I Set the slave camera to stand alone (Inter Cam Sync Mode 0), the IR Seems to work without any problems:
[image]

Can it be that the Slave get slightly out of sync due to it being controlled by the Master Camera. 
I do need the Exposure time, Laser Power and Gain to be like this. And in the future I will need the Camera Sync. 

Basicly there are 2 issues:
In Slave Mode the IR Laser is not visible
When selecting Laser Always On, the laser is flashing instead of stable on. 

This problem occurs on multiple test setups, Always with the slave not showing an IR laser
Is this a problem in the Realsense Firmware, or am I configuring something wrong?

MartyG-RealSense, 2021-12-22:
Hi @PorschefanRoel  In https://github.com/IntelRealSense/librealsense/issues/8129#issuecomment-756796426 a RealSense team member advises that because the D435's projector pulses in line with exposure, if there are two D435 that are not currently successfully hardware synced then sometimes their pulses will be on at same time and sometimes not.  It is suggested to set both cameras' projectors to Always On or use hardware sync (which you have been attempting to do).

Can you confirm please whether you have created a **hardware sync cable** to join the two cameras together using their external hardware sync pins that are underneath a small cover on the top of the camera casing, as described in the hardware sync guide linked to below.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

[image]

Unless the cameras are joined with such a cable, or you are alternatively generating a master 'trigger pulse' signal using external signal generator hardware, then the cameras will not hardware-sync even if Inter Cam Sync Mode settings have been defined.

I also note on the slave camera images that its timestamp is in red text, suggesting that the **software timestamp** (system time) is being used rather than the hardware timestamp on the master camera whose timestamp is in normal white text.

PorschefanRoel, 2021-12-22:
I indeed have a sync cable connected (purle on both sides to pin 5, black to pin 9)
[image]
[image]

To better see it's orientation, this is how the cable looks when connected to the D435:
[image]

I saw that I hadn't Enabled the "Frame Metadata" message when I started the Realsense software. So after enabling the "Frame Metadata", I immediately started the streams (as master & slave), but forgot to set the resolution to my desired 848x480. Then the laser was visible on both images and all timestamps where white:
[image]

Then I stopped the cameras, set it to 848x480, all other settings are unchanged. And the laser was again no longer visible. Times where still shown in white
[image]

When I stop the camera streams again, move back to 640x480, the laser will stay not visible. When I then stop the cameras, close the Realsense Viewer, open the realsense viewer again and start the stream again on 640x480. The laser is visible again.

When I stop the streams, close the Realsense Viewer, Open the realsense viewer, set resolution to 848x480, the laser is not visible. 

During all these runs the timestamps stayed white.

MartyG-RealSense, 2021-12-22:
Thanks @PorschefanRoel  for the very detailed feedback.  If you reboot the computer, launch the Viewer and set both cameras to 848x480, is the laser visible?  If it was then it might suggest that the problem is occurring when the streams are stopped and then started.

PorschefanRoel, 2021-12-22:
And thank you for the fast response.
This are the steps I've done:

Shut Down PC
Startup PC
Startup Realsense Viewer
Now 1 realsense viewer is shown in the bar 
Expand "Stereo Module" menu
Set "Resolution" to 848x480
Set "FPS" to 6
Disable "Auto Exposure"
Expand "Controls"
Set "Exposure" to "1000" 
Set "Laser Power" to "360"
Set "Inter Cam Sync Mode" to 1
Collapse "Stereo Module"
Start Camera
[image]

Master camera is now started

Click "Add Source"
Do the exact same steps as above, until "Inter Cam Sync Mode", have set that to 2.
[image]

But unfortunatly, still no laser on the slave.
[image]

What I did find out was that the runs where the laser was shown on 640x480. The FPS was run on 15FPS, instead of 6, when running on 6FPS and 640x480, there is also no laser shown. The problem thus occurs on multiple resolutions, when run on 6FPS.

MartyG-RealSense, 2021-12-22:
In general I would recommend avoiding 6 FPS and using 15 FPS if possible, as errors as more likely to occur at 6 that disappear at 15.  At 6 FPS, frames may be arriving slowly enough that the pipeline time-outs while waiting for new frames to arrive.

One way to avoid such problems may be to stream at 15 FPS but simulate a lower frequency by only handling every 'nth' frame, such as setting 30 FPS and using only every 5th frame to achieve 6 FPS.  The link below provides more information about this.

https://support.intelrealsense.com/hc/en-us/community/posts/360043865234/comments/360011260594

PorschefanRoel, 2021-12-23:
Due to the current 2.0 cable it isn't possible for us to use the 15FPS at the required 848x480 resolution, so we will keep using the 6FPS for now and turn off the Sync. If that gives good enough results we'll use that, if sync on a later moment seems needed we'll look into upgrading the cables to a 3.0 solution to make the 15FPS (or more) possible.

Thank you for the support

MartyG-RealSense, 2021-12-23:
You are very welcome, @PorschefanRoel - good luck!

MartyG-RealSense, 2021-12-29:
Hi @PorschefanRoel  Do you require further assistance with this case, please?  Thanks!

PorschefanRoel, 2022-01-03:
For now we don't need further assistance. Thanks for the help

MartyG-RealSense, 2022-01-03:
Thanks very much @PorschefanRoel for the update - I will therefore close the case as you do not require further help for the time being.  Thanks again!

closed 2022-01-03 by MartyG-RealSense

PorschefanRoel, 2023-03-21:
I know this is an old issue of mine, but due to synchronisation issues I came back to this. I've now updated the firmware to 5.14.00.00 and using the Realsense Viewer 2.53.1 and at first glance the laser seems to work on both the master as the slave cameras.

MartyG-RealSense, 2023-03-21:
Thanks too for this update  :)